dw-0/kiauh #137: Feature: Whiptail dialogue and internalization support

dingyifei, 2021-11-09:
I'm attempting to rebuild the current kiauh interface with whiptail(supplied by Debian by default) and Gettext on my fork. Whiptail dialogue should be way easier to manage and update. Any thoughts on it? I'm not good at bash so it might be a bad idea...

dingyifei, 2021-11-09:
The main reason behind using whiptail is ditching the 57 char width because it's impossible to get this exact width with gettext localization.

th33xitus, 2021-11-09:
Hi,

generally i think it is a great idea if whiptail comes pre-installed on basically every debian based distro. I think that is very important. In the past i already thought about re-building the UI with something like whiptail or ncurses (i think ncurses is what klippers `make menuconfig` uses). It looks way more professional and clean. I just haven't had the time to dive into that topic and also was a bit intimidated because it seemed rather complex compared to simple regular shell scripting.

> Any thoughts on it? I'm not good at bash so it might be a bad idea...

I really like that you want to start to go down that route. 👍 
I'm not good at bash either, so there are two of us already 😄😛

dingyifei, 2021-11-09:
I'm able to get a functional main and error page. Whiptail's formatting options are ...  not great... On the other hand, raspi-config use whiptail so it is definitely doable

[image]

dingyifei, 2021-11-12:
#138 WIP

th33xitus, 2022-06-04:
Closed due to inactivity. I suspect work on this will not be resumed.

closed 2022-06-04 by th33xitus