dw-0/kiauh #140: Webcamd uses vcgencmd, it would be useful to find out if the script is running on the platform where it is available before using it

wykys, 2021-11-12:


th33xitus, 2021-12-23:
I don't understand. Please elaborate on this issue.

wykys, 2021-12-23:
For example, this command is not available on orangepi ...

th33xitus, 2021-12-23:
And what is it needed for?
Im not the maintainer of the webcamd script or the mjpg-streamer repository.

wykys, 2021-12-23:
I thought you were the author of the `webcamd` script...

I didn't look into the installation code. Only now have we glanced at you downloading the script: https://github.com/th33xitus/kiauh/blob/master/scripts/install_mjpg-streamer.sh#L51
from:
https://github.com/mainsail-crew/MainsailOS/blob/master/src/modules/mjpgstreamer/filesystem/root/usr/local/bin/webcamd

The problem is with the `webcamd` script just mentioned, so the problem probably belongs there.

th33xitus, 2021-12-23:
No that script was created by someone else.
I am not sure if someone will fix it... currently there is some development happening on a project which utilizes for example ustreamer instead of mjpg-streamer. Maybe this works even better for you? This is the project: https://github.com/mainsail-crew/crowsnest

wykys, 2021-12-23:
I'll try, I also ran mjpg-streamer, but I had to edit the webcamd ...

closed 2022-11-05 by th33xitus