dw-0/kiauh #144: Klipper UDS service bug

ils15, 2021-11-16:
Installed Klipper as usual and cannot run correctly

https://i.imgur.com/zaKDm09.png

Please check if the Klipper service is running and an UDS (Unix Domain Socket) is configured.

th33xitus, 2021-11-16:
Can you please describe what do you think the issue with KIAUH is in this situation?

Can you please post the output of the following command, once you logged in via ssh:
`cat /etc/systemd/system/klipper.service`

ils15, 2021-11-16:
Hi, because i used kiauh and no problem
cannot find klipper in this directory

cat: /etc/systemd/system/klipper.service: No such file or directory
But kiauh showing klipper installed 

https://i.imgur.com/RYKlZua.png

ils15, 2021-11-16:
Hi...
Fixed using a old klipper version

closed 2021-11-16 by ils15