dw-0/kiauh #145: Where does Kiauh get the version numbers?

hapklaar, 2021-11-16:
Currently I see an update available for Klipper:

old: v0.10.0-113
new: v0.10.0-152

How can I determine what has been changed from old to new? I can't seem to find any reference to these build numbers anywhere, not even on the klipper github page.

If I need to browse through the commits to see what has changed, would it be possible for Kiauh to show the latest commit for a certain version?

th33xitus, 2021-11-16:
> How can I determine what has been changed from old to new?

You are on v0.10.0 commit 113
Most current is v0.10.0 commit 152
So in your case, look at the last 39 Klipper commits to see what has changed in the meantime.

In the past i displayed truncated commit hashes, but users seem to generally feel less intimidated by the current display method instead of the 7 figure SHA1 i displayed before. And the big pro there is, you can directly see how many commits you are behind. Thats impossible with displaying the shortened SHA1. So that way you get even more info out of the number than before.

hapklaar, 2021-11-16:
One last question: Could I suggest showing the last commit between parenthesis behind the version numbers? That would be even more clear and save us from having to count commits.

th33xitus, 2021-11-16:
The space required in the UI for displaying that would be double compared to now. So with the current state of the UI, i would say no to that i am afraid, im sorry.

hapklaar, 2021-11-16:
Gotcha

closed 2021-11-16 by hapklaar