dw-0/kiauh #148: OctoPrint multi instance sharing plugins

jimmerson17, 2021-12-01:
I used KIAUH to install two Klipper instances and two OctoPrint instances.  I planned to use one to control my CR-6 SE running Marlin, and the other to run a Klipper powered Wanhao i3.

Adding plugins to one installation of Octoprint affected the other.  I installed OctoKlipper to one instance, and it showed up on both.  Is this normal?  

Would I be better served just installing Fluidd on top of OctoPrint, and then only setting up one OctoPrint instance for Marlin and leaving the other one untouched, and then using the Fluidd instance to run my i3?

th33xitus, 2021-12-01:
Okay i see, so it seems plugins are shared between instances because they use the same python virtualenv. That's somewhat unflattering. And so it seems the only solution for that would be, to give each instance its own python virtualenv during installation. I have to test that, it will increase disk usage by a lot and also cluttering of the home directory with even more folders will increase. Im not familiar enough with OctoPrint to know if there is a better approach to that. But i can't think of one right now.

> As a side note, would I be better served just installing Fluidd on top of OctoPrint, and then only setting up one OctoPrint instance for Marlin and leaving the other one untouched, and then using the Fluidd instance to run my i3?

This would indeed be a better approach and the one i would suggest tbh. If you don't need special OctoPrint plugins, Mainsail or Fluidd should be preferred imho. They are tailored to Klipper and use Moonraker to directly communicate with it.

jimmerson17, 2021-12-01:
Good to know.  When I tried to set it up like that originally, I got a warning when installing Fluidd that complained about having OctoPrint already, and that it wanted me to disable the OctoPrint service.  Wouldn't that disable OctoPrint and keep more from using it?

th33xitus, 2021-12-01:
Good point. That's a warning still included "from the early days" were it was unclear yet if an active OctoPrint service can somehow interfere with Moonraker. But i think i can remove that, i haven't come across any issues yet that running Mainsail/Fluidd besides OctoPrint creates problems.

But there is a thing for people using OctoPi and then try to install Fluidd or Mainsail. Because they use NGINX as reverse proxy and OctoPrint uses by default haproxy, KIAUH will ask to either disable or remove lighttp and haproxy. But i don't think you used OctoPi because you said you installed OctoPrint with KIAUH.

jimmerson17, 2021-12-01:
Yeah, I'm not suing OctoPi. I specifically installed just the Pi OS Lite image and KIAUH to avoid any weird overlap.

closed 2022-05-29 by th33xitus