dw-0/kiauh #149: mjpg-streamer needs cmake on Ubuntu

mrbrdo, 2021-12-02:
On a fresh install of ubuntu server 20.x, mjpg-streamer install will fail (maybe shown as successful) because cmake is missing. apt install cmake fixed it.
need to update deps

TodWulff, 2021-12-21:
Yep, came here to report same on Ubuntu 20.04.3 LTS (desktop flavor, in my case).
After the failed install via the script, I used the script to remove the service
And then did `sudo apt-get install cmake`.
A script reinstall thereafter worked.

th33xitus, 2021-12-21:
I'll have to look into it. The dependencies the script checks contain cmake:
https://github.com/th33xitus/kiauh/blob/master/scripts/install_mjpg-streamer.sh#L22
Not quite sure what goes wrong. I try to find out.

th33xitus, 2021-12-24:
I tried reproducing the issue on both, an ubuntu 20.04 server VM and an ubuntu 20.04 desktop VM. I had no issues installing mjpg-streamer or rather a missing cmake error. cmake got installed successfully. Im not completely sure why that is the case for me. Im currently reworking a bit on the develop branch which will be merged into master after some more testing (ubuntu 21.10 server and desktop and RaspberryPi OS Bullseye). Then i have to wait for further error reports if there still exist some or this issue can be seen as rather "fixed"...

xcasxcursex, 2021-12-26:
Saw this on Mint 20.2 also. Was a brand new fresh installation made just for klipper, I'm very sorry I didn't take note of the exact message (I was preoccupied with a paranoid fear of frying my printer with this alien firmware...), but I could see that it was being read from the requirements file and an error suggested it was failing, but it then called it a success and carried on. Pretty sure that the error was coming from whatever the script called to install cmake (apt?), but the script didn't detect that error on return. I saw a post about a need to install cmake manually, did it, and all was well.

Again, sorry I didn't grab the error message. Hopefully just the report of it occurring or one of these little details, might help.

th33xitus, 2022-05-30:
with the release of v4 i tested compiling of mjpg-streamer on ubuntu 20, 21 and 22, server and desktop versions. no issues could be reproduced. in that regard i will close this issue.

closed 2022-05-30 by th33xitus