dw-0/kiauh #154: Moonraker cannot be Installed with SKR Brand Klipper

Heisenberg19x, 2021-12-16:
### What happened

[image]
Klipper is installed at the shown SKR Brand for the BTT SKR Mini E3 V3

### What did you expect to happen

[image]
When trying to Install Moonraker KIAUH says no Klipper Installed

[image]
When trying to Install Klipper KIAUH says Klipper already Installed

### How to reproduce

Install SKR Brand Klipper and try

StefanMeGit, 2021-12-18:
The same happens to me... running KLIPPER on an Ardunie DUE. It looks like its not an SKR Brand Klipper problem. I am using the master branch.

[image]

th33xitus, 2021-12-18:
The brand is not really the problem here.
The issue is, that you both probably have a rather old Klipper installation which means Klipper is installed as an init.d service.
A while ago i changed KIAUH the way that it specifically only looks for systemd service files anymore. And throws that error if none are found. Using the official upstream in that case is no real issue, because one could easily just re-install Klipper with KIAUH and then the service type is the expected one. But installations of a forked/modified Klipper would break because currently KIAUH only installs the official upstream Klipper... no inofficial custom forks.

Im currently thinking about a solution and if it still would be okay to just allow the installation of Moonraker no matter which type of Klipper service exists (technically Moonraker doesn't even need a Klipper service available). In theory i could decouple the Moonraker setup completely from any existing Klipper installation. This would, on the other hand, get rid of some "comfort" as in pre-entering the correct number of needed Moonraker instaces if more than one Klipper instances are found. But at that point i probably should ask myself if limiting installations is worth such little comfort gains in exchange. Probably not.

I try to come up with a solution.

th33xitus, 2021-12-30:
Commit https://github.com/th33xitus/kiauh/commit/4d834db5df781adbb8019b6349730eab22750f37 now "unlinks" the Moonraker installation part of the script completely from Klipper. You are not limited anymore if there is no Klipper installation at all or a very old one (those older init.d services). The only thing that is left is a little part which counts any exisitng Klipper installations (no matter if systemd or init.d) for displaying it to the user simply for informational reasons.

closed 2021-12-30 by th33xitus