dw-0/kiauh #156: Multi printer (Moonraker) detection detects Moonraker telegram bot as a printer

Lupusalix, 2021-12-24:
### What happened

When trying to install a theme for mainsail, the check for multiple printers thinks that the telegram bot is a second printer.

### What did you expect to happen

It shouldn't recognize the telegram bot service configuration as a printer.

### How to reproduce

Install Mainsail, klipper, moonraker, telegram bot, Try to install a mainsail theme.

### Additional information

/scripts/ms_theme_installer.sh
Line 74 `printer_count=$(ls /etc/systemd/system/moonraker*.service | wc -l)` checks true for the telegram bot.
```
ls /etc/systemd/system/moonraker*
/etc/systemd/system/moonraker.service  /etc/systemd/system/moonraker-telegram-bot.service
```

Lupusalix, 2021-12-27:
@th33xitus 
Thank you very much. Works flawlessly :D

th33xitus, 2021-12-27:
@Lupusalix Yw, but keep in mind, that is the develop branch you probably have checked out now. I plan to merge the state of that branch to master in the next days. Then this issue here will auto-close as well.

commit 1f40686e (2021-12-24): fix: improve moonraker service detection (fixes #156)
closed 2021-12-30 by th33xitus