dw-0/kiauh #157: Multiple errors before updating

INeDiA, 2021-12-30:
### Linux Distribution

Raspbian
KIAUH v3.1.0-85

### What happened

Immediately after pressing _2_ (update) I get this series of errors:

```
error: object file .git/objects/0a/541bbea1abbef1278a4527de40e65e9da1869f is empty
error: object file .git/objects/0a/541bbea1abbef1278a4527de40e65e9da1869f is empty
fatal: loose object 0a541bbea1abbef1278a4527de40e65e9da1869f (stored in .git/objects/0a/541bbea1abbef1278a4527de40e65e9da1869f) is corrupt
fatal: the remote end hung up unexpectedly
error: object file .git/objects/0a/541bbea1abbef1278a4527de40e65e9da1869f is empty
error: object file .git/objects/0a/541bbea1abbef1278a4527de40e65e9da1869f is empty
fatal: loose object 0a541bbea1abbef1278a4527de40e65e9da1869f (stored in .git/objects/0a/541bbea1abbef1278a4527de40e65e9da1869f) is corrupt
```

then the regular menu shows up; I didn't tried to update any of my software, I'm just afraid this issue can break my system.

### What did you expect to happen

No errors :)

### How to reproduce

1) Start kiauh
2) Press 2 to go to the update menu

Any other menu works nice.

th33xitus, 2021-12-31:
Hi, please see https://github.com/th33xitus/kiauh/issues/153

closed 2021-12-31 by INeDiA

INeDiA, 2021-12-31:
removing then reinstalling moonraker fixed the issue
sorry to bother!

th33xitus, 2021-12-31:
Don't worry, no problem!
I wonder what the root cause of such issues is. And i wonder if i can somehow "auto fix" that. We'll see...

INeDiA, 2021-12-31:
A little update: I fixed the issue with kiahu, meaning that I don't see any error anymore in the script, but moonraker install has been compromised.
I'm trying to fix it, also with the help of the owner of moonraker project, but it's still not working (and I can't print).
My last resort will be a complete wipe of the system :(