dw-0/kiauh #158: nginx IPv6 configuration

dxgldotorg, 2022-01-02:
### Is your feature request related to a problem? Please describe

Upon attempting to connect to Mainsail Chrome returned a Connection Refused error.  Looking at the open ports in netstat I saw that ports 80 and 81 were bound to the IPv4 0.0.0.0 but not to the IPv6 :: address.  Checked the nginx configuration files and it only had the listen 80 on the mainsail config and listen 81 on the fluidd config.

### Describe the solution you'd like

Add to the nginx config files the following:
For the mainsail config file add after the first listen directive:
`listen [::]:80 ipv6only=on;`

For the fluidd config file add again after the first listen directive:
`listen [::]:81 ipv6only=on;`

Port numbers should of course be substituted if KIAUH is configured with alternate ports.

th33xitus, 2022-01-03:
Yes, i think i can re-add that.
Regarding to the NGINX docs, that `ipv6only=on` flag isn't needed, because it seems to be on by default: https://nginx.org/en/docs/http/ngx_http_core_module.html#listen

closed 2022-01-08 by th33xitus

dxgldotorg, 2022-01-08:
I just found out it was causing connection errors between mainsail and moonraker due to it not detecting being in a trusted IP range.

th33xitus, 2022-01-09:
But did you add the correct ipv6 address to the trusted clients in the moonraker.conf? That's necessary of course...

dxgldotorg, 2022-01-11:
Since my IPv6 addresses are derived from my ISP address and thus subject to change I just removed the IPv6 support once again from the nginx config.  Only the v4 addresses are truly static on my network.

th33xitus, 2022-01-11:
Okay, i think i will revert that change and remove the IPV6 line again. There were no real issues the whole time and in this case i didn't knew it will in turn cause for some unexpected trouble.

commit 7e9d18b5 (2022-01-08): script: add webui ipv6 listen directive (fixes #158)