dw-0/kiauh #159: Option for read-only filesystem

FlatEarthDave, 2022-01-02:
### Is your feature request related to a problem? Please describe

For those who power their RPi's with their printers, it can be a problem when the power goes out, or just turning off the printer, because the SD card can get corrupted if it is writing data to it at the time of power disruption, which happens a lot to me, even when using an SD card provided by The Pi Shop. There's got to be a way to prevent it.

### Describe the solution you'd like

possibly a solution that gives the user the option that enables read-only mode, but disables it for updates and it will store the current gcode file and logs in RAM. It wouldn't persist when the power gets turned off, and resume print wouldn't really work either, but it's just an idea to start with.

### Describe alternatives you've considered

using data=journal mode

th33xitus, 2022-01-03:
I don't think it really matters what kind of SD card you use, expensive or cheap ones. I think all cards are prone to get corrupted on sudden powerloss in one way or the other. 

> There's got to be a way to prevent it.

There is indeed a way 😄 Honestly, i would rather steer people into the direction of just shutting down their Pi correctly before they unplug the power -> `sudo shutdown now` or `sudo shutdown`

philippejadin, 2022-01-09:
Would it helps to boot from a usb memory instead of sd card? I'd tend to say yes but tests would need to be done.

th33xitus, 2022-01-16:
So i decided that this feature don't really fit in the scope of KIAUH. I rather prefer to stick to basic installation helper scripts of components directly revolving around Klipper. I don't think KIAUH should support modifying systems to make them read-only or anything similar. I expect users who really want to go that route of having a system set up in a very special way to do it on their own. And in case of the initial request, i think with a proper shutdown you might extend the lifespan of a common SD card by a lot.

Im not going to implement something like this. Sorry.

closed 2022-01-16 by th33xitus