dw-0/kiauh #160: Not connecting to MCU on SD-flash-method and UART

Tomblarom, 2022-01-04:
### Linux Distribution

```
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 10 (buster)
Release:        10
Codename:       buster
Linux rpi3-3d-ender 5.10.60-v7+
```

### What happened

When flashing via SD-Card Update in combination with a UART-connected MCU, no connection will be established.

```
Flash:         2) Updating via SD-Card Update
Connection:    2) UART
MCU:           1) /dev/ttyAMA0
Board type:   15) btt-skr-turbo-v1.4
Baud rate:    250000
```

After a timeout, I get this error:

```
SD Card Flash Error: Unable to connect to MCU
Traceback (most recent call last):
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1172, in main
    spiflash.run()
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1121, in run
    self.run_reactor_task(self.run_reset)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1108, in run_reactor_task
    k_reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 310, in _dispatch_loop
    timeout = self._check_timers(eventtime, busy)
  File "/home/pi/klipper/klippy/reactor.py", line 156, in _check_timers
    t.waketime = waketime = t.callback(eventtime)
  File "/home/pi/klipper/klippy/reactor.py", line 48, in invoke
    res = self.callback(eventtime)
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 1074, in run_reset
    self.mcu_conn.connect()
  File "/home/pi/klipper/scripts/spi_flash/spi_flash.py", line 813, in connect
    raise SPIFlashError("Unable to connect to MCU")
SPIFlashError: Unable to connect to MCU
```

### What did you expect to happen

--

### How to reproduce

--

Tomblarom, 2022-01-04:
**Solution:** Since Klipper is blocking the access, it has to be stopped manually beforehand using `sudo service klipper stop`.

th33xitus, 2022-01-04:
Odd... the klipper service should get stopped with this line: https://github.com/th33xitus/kiauh/blob/master/scripts/flash_mcu.sh#L210
Im not quite sure why that wasn't the case for you. I have to investigate that. But i can't recreate the exact scenario, i don't have any UART connected MCUs anymore.

Tomblarom, 2022-01-04:
I assume it's the case. There's a `/etc/init.d/klipper` file.

th33xitus, 2022-01-04:
Okay seems like the `sudo systemctl stop klipper` doesn't work in that case. I have to find a workaround for that. Thanks for reporting. (You might want to consider re-installing Klipper if possible, so the old SysV init service gets replaced by a more modern Systemd service)

closed 2022-05-28 by th33xitus