dw-0/kiauh #162: Use 'q' to quit kiauh from any menu

cbc02009, 2022-01-08:
### Is your feature request related to a problem? Please describe

Nope, this is not related to a problem other than me having to go through the immense effort to press one additional key.

### Describe the solution you'd like

It would be nice if more menus recognized q as a quit kiauh option.

### Describe alternatives you've considered

I could just keep entering b until I return to the main menu.

### Additional information

I understand this is very low priority. Just thought it would be useful.

th33xitus, 2022-01-10:
To understand you correct i need an answer to the following question:
Do you want to directly quit KIAUH at all (so exiting the program) or just quickly get back to the main menu (stay in KIAUH)?

cbc02009, 2022-01-12:
I would like to directly quit KIAUH from menus like the install menu without having the hit 'b' to go back to the main menu first.

th33xitus, 2022-01-12:
You can quit KIAUH at any point in time with pressing `CTRL+C` if your intent is to close it anyways.
It's a general shortcut to break out of a command or process in a terminal.

cbc02009, 2022-01-12:
I could have sworn I tried ctrl +c and it didn't work. Of course retesting it now works fine, so it must have been user error...

closed 2022-01-12 by cbc02009