dw-0/kiauh #163: Kiauh modifying klipper.service file

J89eu, 2022-01-10:
### Linux Distribution

Fluiddpi

### What happened

Insatlled Kiauh and proceeded to install Mainsail using a different port. Upon completion I was unable to start Klippy. After checking files found that klipper.service had been modified and klippy.py was not present so python was trying to rubn the klipper.cfg file.

[image]

### What did you expect to happen

Klipper.service unaffected.

### How to reproduce

For me, I had FluiddPi image and then installed Minsail on port 81 and KlipperScreen.

th33xitus, 2022-01-10:
Neither the Mainsail nor the KlipperScreen installer touch the Klipper service file.
So you are leaving out some crucial and very needed information to give any answer to this issue at all and with the current information given to the reproduction steps i can pretty sure tell, that i won't be able to reproduce that issue.

J89eu, 2022-01-10:
In that case just close off the issue as I cannot provide any more info as that was all I have done.

Thanks

J89eu, 2022-01-10:
Hi, I have reproduced the error on my other printer. It breaks when you install the Theme Manager for Mainsail.

th33xitus, 2022-01-10:
There is no such thing as "installing the theme manager".

Okay so i have one assumption:
Do you maybe get a warning, that you need to set a Klipper configuration directory? And then you are asked to specify a folder?

J89eu, 2022-01-10:
[image]
This thing then, there's no need for the condescending tone when I took the effort to log this report despite not being the only one facing the issue according to the Voron discord. Theme insatller, theme manager, hardly much of a difference, you must have know what I was talking about. And yes, I matched my directory on both servers.

th33xitus, 2022-01-10:
It wasn't my intention to sound condescending. But it is very important to me that i get very detailed explanations on what has been done and what errors came up. And telling you that there is no such thing as INSTALLING the theme manager was just meant to explain to you that there is nothing installed. Of course i know what you were talking about. I just wanted to get that straight. Because there is no manager getting installed, the tool itself is the manager if you will. Again, wasn't meant to sound condescending at all, i have no reason for that.

May i ask what exactly did you type in, when asked what your configuration directory is? Because this can probably lead to the root of the issue.

th33xitus, 2022-01-10:
Okay i see, yes that is also the default.
I try to reproduce the issue. Thanks for the info.

p5y, 2022-02-25:
I've reproduced this on two separate fresh installs over the last days:
1. install raspberry OS lite legacy (buster)
2. install git and kiauh
3. install klipper, moonraker, fluidd in that order
4. change klipper config dir location via kiauh
5. check klipper.service, klipper location is stripped

th33xitus, 2022-02-26:
> 4. change klipper config dir location via kiauh

Thank you for reporting this. I was indeed able to reproduce the issue now with THAT kind of information.
If the topic starter also changed the config dir location afterwards, he left out that very cruicial information which would be the reason why i wasn't able to reproduce it until now.

th33xitus, 2022-02-26:
I pushed a fix for the broken function. I will close this issue for now. If the issue is still present though, please open a new issue.

closed 2022-02-26 by th33xitus