dw-0/kiauh #164: Adding additional instance to the existing one

dreamsailorPL, 2022-01-10:
### Is your feature request related to a problem? Please describe

Perhaps I am blind, but I have not seen how to add additional instance of klipper to the existing one.

I have one instance running just fine, but I would like to add 2nd printer.
I know that for fresh install it is possible to choose how many printers will you run, but it is possible to add new one to the existing one?

Many thanks,
Chris

### Describe the solution you'd like

From the KIAUH menu add additional instance.

th33xitus, 2022-01-10:
Hi, please see #130 😄

dreamsailorPL, 2022-01-10:
Thanks, I dont know how I missed that. I bet I have searched the other topics.

closed 2022-01-10 by th33xitus