dw-0/kiauh #170: Switch from mjpg_streamer to ustreamer

kageurufu, 2022-01-27:
### Is your feature request related to a problem? Please describe

mjpg_streamer is terrible

### Describe the solution you'd like

[ustreamer](https://github.com/pikvm/ustreamer) is way better

### Describe alternatives you've considered

If not switch altogether, add ustreamer as a (default) alternative?

th33xitus, 2022-01-27:
Hi, i actually plan to switch to https://github.com/mainsail-crew/crowsnest one day.
I think that would actually also fit your feature request, right?

kageurufu, 2022-01-30:
Yeah, that would totally work!

I've been helping a few other people switch to ustreamer to fix issues,
crowsnest looks like it defaults to ustreamer.

Awesome :)

CODeRUS, 2022-02-13:
just my 5 cents. switch to crownest will break all non-raspberry installations. crownest using raspios specific features and package dependencies. i don't think mainsail team want to adjust crownest to make it fully working on different distros.

th33xitus, 2022-02-13:
@CODeRUS can you explain what the exact points are which make it break? At the moment i am not really familiar with it, since it has rather low priority to get included.

susisstrolch, 2022-02-13:
Main problem is the crows nest install script. 

- hardcoded user ‘pi’ in unit file
- Unconditional removal of existing folders (like mstreamer)
-  dependency on libraspberrypi-dev
- Downloads a binary blob (arm64) for golang

susisstrolch, 2022-02-13:
The dependency isn’t really a problem, because it’s in an aptget block and can get factored out. 
I already started those changes. 
Also the binary blob isn’t a real problem, can get ifdefed based on mcu arch.

CODeRUS, 2022-02-13:
as soon as mainsail crew fix such problems

KwadFan, 2022-02-13:
I will provide, multiarch support, but first I want to get it running perfectly with webrtc on Pis especcially MainsailOS.
Most Parts in the nightly branch are already prepared for that. There some bits left to do.
ex: webcamd.service, libraspberry-dev...

Promised! You x86/amd64 guys will get crownest ported ;)

Regards

KwadFan, 2022-02-14:
So, if some of you x86 guys are crazy enough to test some new stuff:

https://github.com/mainsail-crew/crowsnest/tree/feat/multiarch

This includes a webrtc server but, it isnt implemented in Mainsail yet. Pure Alpha Stuff.

If you setup webrtc as mode,you can reach a simple test page via http://<printerip>:8085/

If you find Bugs or something that does not work please open a Issue on crowsnest repo with [feat-x86] in Header.

Regards

th33xitus, 2023-02-03:
Today, a crowsnest installer got merged into KIAUH. So this issue can be closed and marked as solved.

closed 2023-02-03 by th33xitus