dw-0/kiauh #171: Select python version upon klipper installation

CODeRUS, 2022-02-12:
### Is your feature request related to a problem? Please describe

klipper now supports python3, python2 is EOL, pip installation on Debian 11 and so is PITA

### Describe the solution you'd like

simple -p argument for virtualenv creation command, with python2 as default. can be set to python3, no extra actions required.

th33xitus, 2022-02-15:
Generally i think there should be no issues installing Klipper with Python 3 even by default.

My current idea for this would be a simple dialog asking the user if he wants to install Klipper with python3, in case there is no python2 installed on the operating system.

I haven't heard any issues so far from running Klipper on python3, but on the same hand i am not sure how widespread the current use of python3 with Klipper is. Currently the thing is like this: if you use a distro which does not come with python2 pre-installed anymore (so Debian Bullseye for example as you already mentioned it), the current Klipper install will install python2 on the system (in form of the python-dev package which is python2). Which in turn is kind of useless because it is not required to run Klipper and you could just use python3 instead. The only issue i saw was, that building the firmware requires python2 by default and will result in an error when you try and build the firmware with a simple "make" command. Python2 is hardcoded in the Makefile and, im not sure why exactly, a PR to Klipper with making it less strict was declined lately. Though, you can build it with python3 by using "make PYTHON=python3" instead and it will work just fine. Users just need to know that if they someday decide to build a firmware on their own while not being guided by KAIUH or so. 

Also, i think not all other scripts Klipper comes with are python3 ready yet? I haven't investigated that further and in detail and im not sure which issues will or can occur due to that.

Generally though, i am PRO adding python3 support to KIAUH. I just have to find time to accomplish that and then time to test it at least a bit.

CODeRUS, 2022-02-15:
@th33xitus Kevin mentioned he is doing tests with both python2 and python3 installations: https://github.com/Klipper3d/klipper/pull/3278#issuecomment-932656677

th33xitus, 2022-02-15:
We have had quite an interesting discussion going on in the Klipper discord regarding this. The real "problem" will be, to keep old installations functioning and not break those installs. Thats of course something what i have to keep in mind in case i decide to tackle this feature in the future. Though, i have no ETA when i will be able to implement something like that. I have quite much going on at the moment, in other projects and in Real Life.

CODeRUS, 2022-02-15:
as for kiauh it have no difference if python2 or python3 is used inside virtualenv. it's already working if one create virtualenv with python3. klipper updates installed with no problem, etc. klipper is launched using env/bin/python which have no versioning, same for env/bin/pip. I'm already using python3 with no problems, and kiauh have no modifications at all.

th33xitus, 2022-02-15:
Correct. What about updating the firmware on your MCU?

CODeRUS, 2022-02-15:
@th33xitus well, i have python2 installation indeed. but did not tried to flash mcu yet, can test it soon with PYTHON parameter

CODeRUS, 2022-02-16:
just flashed stm32 board with pytohn3 just fine

DragonFlyer007, 2022-03-18:
Hope to see this happend soon. Really like Kiauh, but need to get a fresh dual klipper install with Py3.

closed 2022-05-29 by th33xitus