dw-0/kiauh #173: Failed building wheel for pycairo on Klipperscreen

darkpoetry, 2022-02-15:
### Linux Distribution

Debian 11.2 aarch64

### What happened

Fails to install KlipperScreen but thinks it was successful. Error message below

### What did you expect to happen

Should install without errors.

After manually upgrading pip, I get a slightly different error:

  Building wheel for pycairo (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for pycairo (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [6 lines of output]
      running bdist_wheel
      running build
      running build_py
      running build_ext
      'pkg-config' not found.
      Command ['pkg-config', '--print-errors', '--exists', 'cairo >= 1.15.10']
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for pycairo
Failed to build pycairo
ERROR: Could not build wheels for pycairo, which is required to install pyproject.toml-based projects
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.
>>>>>> KlipperScreen successfully installed!

### How to reproduce

Attempt to install KlipperScreen on v3.1.0-95 in a clean build environment.

### Additional information

WARNING: You are using pip version 21.3.1; however, version 22.0.3 is available.
You should consider upgrading via the '/home/pi/.KlipperScreen-env/bin/python -m pip install --upgrade pip' command.
Requirement already satisfied: websocket-client==1.2.3 in ./.KlipperScreen-env/lib/python3.9/site-packages (1.2.3)
WARNING: You are using pip version 21.3.1; however, version 22.0.3 is available.
You should consider upgrading via the '/home/pi/.KlipperScreen-env/bin/python -m pip install --upgrade pip' command.
Collecting pycairo==1.20.1
  Using cached pycairo-1.20.1.tar.gz (344 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: pycairo
  Building wheel for pycairo (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/.KlipperScreen-env/bin/python /home/pi/.KlipperScreen-env/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /tmp/tmpdwv8r0bt
       cwd: /tmp/pip-install-eq72cfy_/pycairo_4c8472521d37404caabf71e59739b18a
  Complete output (6 lines):
  running bdist_wheel
  running build
  running build_py
  running build_ext
  'pkg-config' not found.
  Command ['pkg-config', '--print-errors', '--exists', 'cairo >= 1.15.10']
  ----------------------------------------
  ERROR: Failed building wheel for pycairo
Failed to build pycairo
ERROR: Could not build wheels for pycairo, which is required to install pyproject.toml-based projects
WARNING: You are using pip version 21.3.1; however, version 22.0.3 is available.
You should consider upgrading via the '/home/pi/.KlipperScreen-env/bin/python -m pip install --upgrade pip' command.
Error: pip install exited with status code 0
Unable to install dependencies, aborting install.
>>>>>> KlipperScreen successfully installed!

###### Restarting KlipperScreen Service ...
Failed to restart KlipperScreen.service: Unit KlipperScreen.service not found

th33xitus, 2022-02-15:
I agree, that error handling in KIAUH is not good and success messages appear even if errors occured, but the root cause of that error doesn't seem KIAUH related. Please consider asking in the KlipperScreen repostiory and report the issue there.

closed 2022-02-15 by th33xitus

darkpoetry, 2022-02-15:
Kiauh is supposed to install the necessary packages through to successfully build these things right? The following were not installed by Kiauh but are required by this program:

libcairo2-dev python3-gi libgirepository1.0-dev

It also appears that the requested xserver-xorg-video-fbturbo is not in the standard repositories but it seems to be unnecessary because it installs without it.

th33xitus, 2022-02-15:
> Kiauh is supposed to install the necessary packages through to successfully build these things right?

In case of KlipperScreen it is a bit different. In that case, KIAUH is more or less just a "frontend" to clone the KlipperScreen repo and then executes its own install script:
https://github.com/th33xitus/kiauh/blob/master/scripts/install_klipperscreen.sh#L37

Everything happening AFTER the cloning of the repo and invoking of the install script is in the scope of the KlipperScreen maintainer.