dw-0/kiauh #174: functions.sh too many arguments

reefwarrior, 2022-02-25:
### Linux Distribution

Linux switchwire 5.13.0-sun50iw9 #2.1.6 SMP Mon Dec 6 13:58:28 CST 2021 aarch64 aarch64 aarch64 GNU/Linux

### What happened

/home/orangepi/kiauh/scripts/functions.sh: line 398: [: too many arguments
/home/orangepi/kiauh/scripts/functions.sh: line 401: [: too many arguments
/home/orangepi/kiauh/scripts/functions.sh: line 404: [: too many arguments
/home/orangepi/kiauh/scripts/functions.sh: line 407: [: too many arguments
/home/orangepi/kiauh/scripts/functions.sh: line 410: [: too many arguments
/home/orangepi/kiauh/scripts/functions.sh: line 414: [: too many arguments

### What did you expect to happen

Should not give errors

### How to reproduce

simply ran:

./kiauh/kiauh.sh

### Additional information

apt-get update and apt-get upgrade already done - no problems seen on screen with regards to update/upgrade.

th33xitus, 2022-03-05:
What Linux Distro do you use? There is no Linux distro called "switchwire".

Im not able to reproduce this.

reefwarrior, 2022-03-05:
Hi! I've (for some reason) been able to overcome this - I THINK that was Armbian - but it's not fully supported by the orangepi, so running ubuntu server at this stage and it went through without any issues! Please mark this as solved? Sorry for wasting your time!

th33xitus, 2022-03-05:
Okay then i count this as an armbian quirk 😄 that's something i also can't test because i don't have any hardware that can run armbian.

> Sorry for wasting your time!

No worries and thank you for your fast feedback! I will close this issue then.

closed 2022-03-05 by th33xitus