dw-0/kiauh #176: mjpg-streamer don't start

NaMoUrGCh, 2022-03-05:
### Linux Distribution

Raspberry PI OS 64bit

### What happened

After successfully install Klipper, Moonraker and Fluidd I have no image from my usb webcam.
Mjgp-streamer was installed with Fluidd install.

It seems to have an error in /var/log/webcamd.log : 

> Starting up webcamDaemon...
>
>--- Configuration: ----------------------------
>cfg_file:      /home/pi/klipper_config/webcam.txt
>camera:        usb
>usb options:   -r 640x480 -f 10
>raspi options: -fps 10
>http options:  -w ./www-mjpgstreamer -n
>
>Explicitly USB device:
>-----------------------------------------------
>
>Found video devices:
>/dev/video0
>/dev/video1
>/dev/video10
>/dev/video11
>/dev/video12
>/dev/video13
>/dev/video14
>/dev/video15
>/dev/video16
>/dev/video18
>/dev/video20
>/dev/video21
>/dev/video22
>/dev/video23
>config file='/home/pi/klipper_config/webcam.txt':USB device was not set in options, start MJPG-streamer with the first >found video device: /dev/video0
><13>Mar  5 18:07:17 pi: Starting USB webcam
>Checking for VL805 (Raspberry Pi 4)...
>  - It seems that you don't have VL805 (Raspberry Pi 4).
>    There should be no problems with USB (a.k.a. select() timeout)
>Running ./mjpg_streamer -o output_http.so -w ./www-mjpgstreamer -n -i input_uvc.so -r 640x480 -f 10 -d /dev/video0
>/usr/local/bin/webcamd: line 151: ./mjpg_streamer: No such file or directory
>Done bring up all configured video device
>
>Goodbye...

Result of lsusb : 
>Bus 002 Device 002: ID 174c:0862 ASMedia Technology Inc. X862
>Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
>Bus 001 Device 003: ID 1908:2311 GEMBIRD Generic UVC 1.00 camera [AppoTech AX2311]
>Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
>Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
ID 1908:2311 is my webcam

On webcam.txt there is only one uncommented line : 
>camera="usb"

### What did you expect to happen

Working mjpg-streamer

### How to reproduce

Just install Killper, Moonraker and Fluidd

th33xitus, 2022-03-05:
Im not an expert with debugging webcam or mjpg-streamer related issues.

I see this line in your webcamd.log:
```
/usr/local/bin/webcamd: line 151: ./mjpg_streamer: No such file or directory
```
Does the folder mjpg-streamer exists in your home directory?
`~/mjpg-streamer`

NaMoUrGCh, 2022-03-05:
Thanks for your quick reply, yes folder ~/mjpg-streamer exist.

th33xitus, 2022-03-06:
And inside of that folder is another folder called `mjpg_streamer` ? (now with an underscore)
The log suggests, that this folder is missing.

NaMoUrGCh, 2022-03-06:
This file is under the sub directory _build

th33xitus, 2022-03-06:
So it is NOT in `~/mjpg-streamer` but in `~/mjpg-streamer/_build` only?
What is the content of `~/mjpg-streamer`

NaMoUrGCh, 2022-03-15:
Sorry for the late reply, I was away of my RPi.

There is the content of the directory
>pi@hypercube:~/mjpg-streamer $ ls -al
total 144
drwxr-xr-x 10 pi pi  4096 Mar  5 17:37 .
drwxr-xr-x 16 pi pi  4096 Mar  6 10:43 ..
drwxr-xr-x  4 pi pi  4096 Mar  5 17:37 _build
drwxr-xr-x  2 pi pi  4096 Mar  5 17:37 cmake
-rw-r--r--  1 pi pi  2701 Mar  5 17:37 CMakeLists.txt
-rw-r--r--  1 pi pi   596 Mar  5 17:37 Dockerfile
-rw-r--r--  1 pi pi   116 Mar  5 17:37 docker-start.sh
drwxr-xr-x  8 pi pi  4096 Mar  5 17:37 .git
-rw-r--r--  1 pi pi   241 Mar  5 17:37 .gitignore
-rw-r--r--  1 pi pi 17987 Mar  5 17:37 LICENSE
-rwxr-xr-x  1 pi pi   510 Mar  5 17:37 makedeb.sh
-rw-r--r--  1 pi pi   772 Mar  5 17:37 Makefile
-rw-r--r--  1 pi pi 15782 Mar  5 17:37 mjpg_streamer.c
drwxr-xr-x  2 pi pi  4096 Mar  5 17:37 mjpg-streamer-experimental
-rw-r--r--  1 pi pi  3695 Mar  5 17:37 mjpg_streamer.h
-rw-r--r--  1 pi pi   285 Mar  5 17:37 mjpg_streamer@.service
drwxr-xr-x 17 pi pi  4096 Mar  5 17:37 plugins
-rw-r--r--  1 pi pi    57 Mar  5 17:37 postinstall.sh
-rw-r--r--  1 pi pi  1350 Mar  5 17:37 README.md
drwxr-xr-x  2 pi pi  4096 Mar  5 17:37 scripts
-rwxr-xr-x  1 pi pi  5117 Mar  5 17:37 start.sh
-rw-r--r--  1 pi pi   330 Mar  5 17:37 TODO
-rw-r--r--  1 pi pi  4659 Mar  5 17:37 utils.c
-rw-r--r--  1 pi pi  5137 Mar  5 17:37 utils.h
drwxr-xr-x  2 pi pi  4096 Mar  5 17:37 www
drwxr-xr-x  2 pi pi  4096 Mar  5 17:37 www-mjpgstreamer

It seems I missing input_* and output_* file

th33xitus, 2022-03-15:
> It seems I missing input_* and output_* file

And you are indeed missing the file `mjpg_streamer`, which is mentioned in the error message.
Im not sure why you also missing the other files. My first suggestion would be to try and re-install mjpg-streamer. Maybe something went wrong during compilation.

NaMoUrGCh, 2022-03-15:
I already try to re-install mjpg_streamer.
The missing file are in the son directory, I will try to make symlink.

NaMoUrGCh, 2022-03-15:
Files input_uvc.so and all output_* are missing, maybe it's a clue

th33xitus, 2022-03-15:
Not a clue for me.

NaMoUrGCh, 2022-03-15:
There is some error at the compilation

[mjpg_streamer_compilation.txt](https://github.com/th33xitus/kiauh/files/8254589/mjpg_streamer_compilation.txt)

th33xitus, 2022-03-15:
I can't help you with that. I assume its due to your OS being 64bit (there is pretty much no gain in using 64bit OS btw)

NaMoUrGCh, 2022-03-15:
Yes, seems that MMAL has lot of trouble in 64bit OS.

closed 2022-04-11 by th33xitus