dw-0/kiauh #177: Editing moonraker.conf with multi instance install tries to restart moonraker.service

ssendev, 2022-03-05:
### Linux Distribution

Debian Buster (Raspberry PI)

### What happened

```
2022-03-05 19:14:14,728 [app.py:log_request()] - 201 POST /server/files/upload (192.168.44.46) [_TRUSTED_USER_] 13.61ms
2022-03-05 19:14:14,768 [base_events.py:default_exception_handler()] - Task exception was never retrieved
future: <Task finished name='Task-9330' coro=<Machine.do_service_action() done, defined at /home/pi/moonraker/moonraker/components/machine.py:159> exception=DBusError('Unit moonraker.service not found.')>
Traceback (most recent call last):
  File "/home/pi/moonraker/moonraker/components/machine.py", line 163, in do_service_action
    await self.sys_provider.do_service_action(action, service_name)
  File "/home/pi/moonraker/moonraker/components/machine.py", line 596, in do_service_action
    await mgr.call_restart_unit(                        # type: ignore
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/dbus_next/aio/proxy_object.py", line 92, in method_fn
    BaseProxyInterface._check_method_return(msg, intr_method.out_signature)
  File "/home/pi/moonraker-env/lib/python3.9/site-packages/dbus_next/proxy_object.py", line 62, in _check_method_return
    raise DBusError._from_message(msg)
dbus_next.errors.DBusError: Unit moonraker.service not found.
```

### What did you expect to happen

`moonraker-1.service` to restart

### How to reproduce

Edit moonraker.conf on a multi instance installation and then hit "Save & Restart" in mainsail

th33xitus, 2022-03-05:
This is not a bug, this is "sadly" the way we have to deal with it currently. You have to restart the Moonraker services on your own via Mainsail's top right power menu.

ssendev, 2022-03-05:
But changing `printer.cfg` restarts the correct klipper instance how does that work? and couldn't the same technique be used for moonraker?

If it's not solvable in kiauh where could it be solved?

th33xitus, 2022-03-05:
> But changing printer.cfg restarts the correct klipper instance how does that work?

Oh it does?
In that case, i have to check what is going on in Mainsail, and if it is something that has to be changed in Mainsail then.

th33xitus, 2022-03-05:
Alright, i talked with @meteyou and we will integrate something in Mainsail to make this possible in a future update.
I'll keep you updated!

th33xitus, 2022-03-05:
The feature will be included in the next Mainsail release: https://github.com/mainsail-crew/mainsail/pull/700
It will be possible to specify which Moonraker instance can restart which Moonraker service, which will cover the requested case.

closed 2022-03-05 by th33xitus

ssendev, 2022-03-06:
Nice. Thanks for the quick response and fix.