dw-0/kiauh #179: Autocommit crash ...

themacboy, 2022-03-21:
### Linux Distribution

Raspberry OS

### What happened

From mainsail I sent a macro that calls autommit.sh script. But then all klipper shutdown.

Looking in the klippy.log I found that error:

Running Command {autocommit}...:
Unhandled exception during run
Traceback (most recent call last):
  File "/home/pi/klipper/klippy/klippy.py", line 201, in run
    self.reactor.run()
  File "/home/pi/klipper/klippy/reactor.py", line 269, in run
    g_next.switch()
  File "/home/pi/klipper/klippy/reactor.py", line 316, in _dispatch_loop
    self._fds[fd](eventtime)
  File "/home/pi/klipper/klippy/extras/gcode_shell_command.py", line 35, in _process_output
    data = self.partial_output + data.decode()
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 10: ordinal not in range(128)
Transition to shutdown state: Unhandled exception during run

### What did you expect to happen

run a back  up to my github account.

### How to reproduce

always on my side, other people is not having that issue. 

### Additional information

Not sure what erro means, looks like some files is incorrectly coded? But not sure how to know the worng file.

Thx in advance !

th33xitus, 2022-03-21:
First of all, that doesn't sound like an actual issue with KIAUH.
And if other people aren't able to reproduce that issue, then it is very likely a self induced error on your side and you have to troubleshoot or debug it. You should consider asking for help on the official Klipper Discord.

closed 2022-03-21 by th33xitus

themacboy, 2022-03-21:
Im agree.