dw-0/kiauh #182: Mainsail on Ubuntu 21.10 possible fix

5ilver, 2022-03-28:
### Linux Distribution

Ubuntu 21.10

### What happened

mainsail failed to display in a browser showing a 403 error

### What did you expect to happen

expected to see mainsail UI

### How to reproduce

Install normally on ubuntu 21.10 and encounter this bug as mentioned in the changelog

### Additional information

it works if the nginx user in nginx.conf is changed to the same user that installed mainsail etc

marc-gist, 2022-03-30:
Also works to fix on Armbian install where you get a 403 until changing nginx user.

PeggyFree, 2022-04-24:
@marc-gist  thank you for the workaround, I was getting crazy with this problem. Also waiting for a real fix.

th33xitus, 2022-05-26:
I currently try to tackle that issue. But to be honest i don't like the method of changing the nginx user from `www-data` to the current username. This seems unsafe to me, as you open your full homedirectory to nginx i assume?
Same goes for just adding `www-data` to the users usergroup. That seems to work, but again sounds unsafe to me, as again you giving access to the full home directory to `www-data`.

Im really not sure what would be a good and safe approach here right now :/

th33xitus, 2022-05-27:
i think i found a solution that is actually much more simpler (and safer) than fiddling around with groups and usernames. so version 4 should actually provide a sufficient fix to that issue.

closed 2022-05-29 by th33xitus