dw-0/kiauh #185: Possible issue after moonraker "update_manager" update?

INeDiA, 2022-04-06:
### Linux Distribution

Raspbian GNU/Linux 11 (bullseye)

### What happened

Updated moonraker via kiahu; after reloading the frontend, fluidd suggested to modify moonraker config due to some deprecated commands.
Did as expected:
```
[update_manager fluidd]
type: web
channel: beta
```
After this the update menu in kiahu shows neither local nor remote version for fluidd; if I try to reinstall/update it, the log shows:
```
###### Downloading Fluidd  ...
http://: Invalid host name.
```

### What did you expect to happen

Install/update fluidd, mainsail, KlipperScreen

### How to reproduce

Update moonraker and modify the config according to [changes to update manager](https://github.com/Arksine/moonraker/issues/410)

th33xitus, 2022-04-06:
KIAUH and the Moonraker update manager aren't linked **at all* and work completely independent of each other.
It doesn't matter what is inside of your moonraker.conf, what is configured and what is not. So this is not the issue.
This also applies to displaying of the versions. The local version is read from a .version file inside of the Fluidd folder. The remote version is read from the github API of the fluidd repo. 

The download URL looks odd, but i wasn't able to reproduce this issue.
Something important is missing here... did you do something else you forgot to mention?

INeDiA, 2022-04-06:
Oops sorry, didn't know about that.
No, I didn't update/modify anything at all, just tried the regular update.

The only weird thing that happened is that github asked my email and name, so I had to
`git config --global user.email "myemail"`
So maybe this is something related with git..?
This would also explain a minor weird thing: [KlipperScreen] is v0.2.1-16 on local and v0.2.1-14 on remote, I'm in the future :D

th33xitus, 2022-04-06:
What versions does the Moonraker update manager show you in Mainsail/Fluidd?
Git asking for credentials is really a weird thing, i saw this a few times now, i still don't know why this is happening. But this should also not be related with your ongoing issue.

Btw, what version if KIAUH are you using?
Maybe a fresh clone of KIAUH will help, for the moment i am really out of ideas :D

INeDiA, 2022-04-06:
In the end I didn't do a fresh install (or anything at all) and just gave it another try: it installed Fluidd withouth any issues.
I didn't even rebooted the pi, I'm really confused.