dw-0/kiauh #192: KIAUH gets stuck on any installation while resolving deltas

Keybored02, 2022-04-24:
### Linux Distribution

DietPi Bullseye

### What happened

I can clone the KIAUH repository without issues. While installing Klipper or Moonraker, however, the installation gets stuck while resolving deltas. It solves 1% (232/23172) and then terminates the SSH session. It's not a git or network issue. I can clone and unpack other repositories just fine. The git check shows no pull request pending, all files match. A fresh OS image doesn't help either. I was able to get it to work previously. 

### What did you expect to happen

To solve all the deltas and carry on with the installation.

### How to reproduce

Follow the instructions on the git readme. No additional steps taken.

Keybored02, 2022-04-25:
Additional info: tried on Armbian, git 2.34.5. Same issue. Changed SSH server and client too. I can clone the Klipper repository manually, outside of kiauh.sh. The KIAUH script shouldn't interfere with that, but apparently, it does. I set up an Ubuntu 20.4 VM under Windows, and everything works fine. No idea why it doesn't under the other two.

th33xitus, 2022-04-27:
Let's keep this issue open for now. Although i haven't looked into it, i can't really imagine the script interfering with the cloning process. Im currently working on a version 4 of the script, and testing on DietPi will be involved, thats when i will try and reproduce this issue.

th33xitus, 2022-05-28:
Just did some testing with the v4-rc branch. No issues during cloning so far. I will mark this issue as "solved" for now.

closed 2022-05-28 by th33xitus