dw-0/kiauh #195: Klipper install fails on Ubuntu 22.04

cbc02009, 2022-05-02:
### Linux Distribution

Ubuntu 22.04 server aarch64

### What happened

Fresh install, cloned kiauh and attempted to install klipper:
```
/=======================================================\
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/

###### Initializing Klipper installation ...

/=======================================================\
|                    !!! WARNING !!!                    |
|        No Klipper configuration directory set!        |
|-------------------------------------------------------|
|  Before we can continue, KIAUH needs to know where    |
|  you want your printer configuration to be.           |
|                                                       |
|  Please specify a folder where your Klipper configu-  |
|  ration is stored or, if you don't have one yet, in   |
|  which it should be saved after the installation.     |
\=======================================================/
/=======================================================\
|  IMPORTANT:                                           |
|  Please enter the new path in the following format:   |
|  /home/chris/your_config_folder                       |
|                                                       |
|  By default 'klipper_config' is recommended!          |
\=======================================================/

###### Please set the Klipper config directory:
/home/chris/klipper_config

###### Set config directory to '/home/chris/klipper_config' ? (Y/n):
###### > Yes

###### Create KIAUH backup directory ...
>>>>>> Directory created!
>>>>>> No config directory found! Skipping backup ...

###### Directory set to '/home/chris/klipper_config'!

>>>>>> Config directory changed!
/=======================================================\
| Please select the number of Klipper instances to set  |
| up. The number of Klipper instances will determine    |
| the amount of printers you can run from this machine. |
|                                                       |
| WARNING: There is no limit on the number of instances |
| you can set up with this script.                      |
\=======================================================/
###### Number of Klipper instances to set up: 1

###### Install 1 instance(s)? (Y/n):
###### > Yes

###### Installing 1 Klipper instance(s) ...

###### Checking for the following dependencies:
● git
>>>>>> Dependencies already met! Continue...

###### Downloading Klipper ...
Cloning into 'klipper'...
remote: Enumerating objects: 30774, done.
remote: Total 30774 (delta 0), reused 0 (delta 0), pack-reused 30774
Receiving objects: 100% (30774/30774), 37.12 MiB | 9.16 MiB/s, done.
Resolving deltas: 100% (23411/23411), done.
Updating files: 100% (1684/1684), done.

###### Download complete!

###### Installing dependencies ...

###### Reading dependencies...
virtualenv
python-dev
libffi-dev
build-essential
libncurses-dev
libusb-dev
avrdude
gcc-avr
binutils-avr
avr-libc
stm32flash
dfu-util
libnewlib-arm-none-eabi
gcc-arm-none-eabi
binutils-arm-none-eabi
libusb-1.0

###### Running apt-get update...
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Reading package lists... Done

###### Installing packages...
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Note, selecting 'libusb-1.0-0-dev' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-doc' for regex 'libusb-1.0'
Note, selecting 'libusb-1.0-0' for regex 'libusb-1.0'
Package python-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  python2-dev python2 python-dev-is-python3

E: Package 'python-dev' has no installation candidate

###### Installing python virtual environment...
/home/chris/kiauh/scripts/install_klipper.sh: line 88: virtualenv: command not found
/home/chris/kiauh/scripts/install_klipper.sh: line 90: /home/chris/klippy-env/bin/pip: No such file or directory
```

### What did you expect to happen

Expected virtualenv to be available.

### How to reproduce

1. Install Ubuntu 22.04 using raspberry pi imager
2. Clone Kiauh
3. Run Kiauh.sh
4. Attempt to install Klipper

### Additional information

package `python-dev` has been renamed to `python2-dev`. 
package `python` has been renamed to `python2`

The binary for `python` has also been renamed to `python2`

Creo2005, 2022-05-08:
solution:

sudo service klipper stop
sudo service moonraker stop
mv ~/klippy-env ~/klippy-env-27
virtualenv -p python3 ~/klippy-env
~/klippy-env/bin/pip install -r ~/klipper/scripts/klippy-requirements.txt
sudo service klipper start
sudo service moonraker start

vkhodygo, 2022-05-28:
This is not a kiauh issue, although it also suffers from `python2` in the scripts.
This is related directly to Klipper and its messy scripts. They should've been updated years ago, but no result so far.

commit 89d38f11 (2022-05-14): refactor(klipper.sh): rename package `python-dev` to `python2-dev` (fixes #195)

On more modern linux distros like Ubuntu 22, the package `python-dev` was renamed to `python2-dev` resulting in `E: Package 'python-dev' has no installation candidate`
closed 2022-05-29 by th33xitus