dw-0/kiauh #196: klipper update bug

mlaci001, 2022-05-09:
### Linux Distribution

RPI Debian Bullseye

### What happened

update klipper:
/home/klipper/klippy-env/bin/pip: line 3: import: command not found
/home/klipper/klippy-env/bin/pip: line 4: import: command not found
/home/klipper/klippy-env/bin/pip: line 5: from: command not found
/home/klipper/klippy-env/bin/pip: line 7: syntax error near unexpected token `('
/home/klipper/klippy-env/bin/pip: line 7: `    sys.argv[0] = re.sub(r'(-script\.pyw|\.exe)?$', '', sys.argv[0])'

### What did you expect to happen

Normally updated klipper.

### How to reproduce

Menu:
2) [Update]
1) [Klipper] 

### Additional information

line 420:     /bin/bash "${KLIPPY_ENV}/bin/pip" install -r "${KLIPPER_DIR}/scripts/klippy-requirements.txt"
I think the "-c" parameter is missing
line 420:     /bin/bash -c "${KLIPPY_ENV}/bin/pip" install -r "${KLIPPER_DIR}/scripts/klippy-requirements.txt"
I

th33xitus, 2022-05-12:
I assume you are not using the current KIAUH master, correct? You checked out the v4-rc branch?

mlaci001, 2022-05-13:
Sorry.
Yes I use the v4-rc branch.
I also saw another error in the update menu under "6) [PrettyGCode]".
[image]

th33xitus, 2022-05-13:
Okay thanks i will have a look.
The issue you mentioned earlier is fixed. I simply reverted to the previous way of calling scripts or commands. So without `/bin/bash` in front of them. Not quite sure why i changed that if it worked flawless lol...

closed 2022-05-29 by th33xitus