dw-0/kiauh #197: Kiauh not working as of latest update

Stickysaurus, 2022-05-13:
### Linux Distribution

Mainsail os

### What happened

Someone walked me through the Kiauh install twice and as of the latest update, it is unstable. Superslicer was unable to start prints, and I was unable to start them manually via the Mainsail ui. After Re-installing mainsail via the Rasberry Pi imager, I am able to print, but am still having issues regarding Superslicer using the most recent version of Mainsail, v2.2.0-beta2. Im sure there is a work-around for this, but as far as plug and play goes, Kiauh is automatically pulling the most recent version of Mainsail which is unstable. 

### What did you expect to happen

I expected Kiauh to download the most recent and stable version of Mainsail, but it is updating straight from the github repository for mainsail's latest releases. 

### How to reproduce

Try a fresh install of mainsail via Kiauh. As of the latest unstable release, it will automatically download it. 

### Additional information

I need sleep

th33xitus, 2022-05-13:
See this for your initial issue of not being able to print: https://github.com/mainsail-crew/mainsail/issues/800

Manually re-install Mainsail stable -> v2.1.2

closed 2022-05-13 by th33xitus

Stickysaurus, 2022-05-13:
> See this for your initial issue of not being able to print: [mainsail-crew/mainsail#800](https://github.com/mainsail-crew/mainsail/issues/800)
> 
> Manually re-install Mainsail stable -> v2.1.2

Yea, theres an issue with manually installing. I will admit I am pretty new and struggled alot with the install. I was working with a few people who are very familiar with Kiauh and have used it multiple times on multiple different printers. Changing the source in the config did not change the source install location, it just kept downloading the beta. I think its an issue with mainsail itself and not an issue with Kiauh. After installing mainsail on the Pi Imager, I have managed to get the most recent version of mainsail to work. I am using the exact config as my friend who has no issues and I am still having issues with the file upload in the ui. I was able to start it, but I had to use the interface on the printer as it wouldn't start automatically. I appreciate all the work everyone has put into this project, and also appreciate and advice someone may have for the issue. Sorry if I am posting this in the wrong section, I am pretty new to github and only created this account to relay this issue. Thanks all

th33xitus, 2022-05-13:
> I think its an issue with mainsail itself and not an issue with Kiauh. 

No, KIAUH downloading the latest Mainsail version, even if it unstable is indeed the case for the moment. I am working on a new major release of KIAUH where this behavior will be changed/can be made configurable.

> Sorry if I am posting this in the wrong section, I am pretty new to github and only created this account to relay this issue.

No problem. I think if you need further advice and help, you should either join the Klipper Discord or the Mainsail Discord (or maybe both, since Mainsail related questions might get answered faster on the Mainsail Discord, and Klipper related ones on the Klipper Discord.)

https://discord.klipper3d.org/
https://discord.gg/mainsail

Stickysaurus, 2022-05-13:
> No, KIAUH downloading the latest Mainsail version, even if it unstable is indeed the case for the moment. I am working on a new major release of KIAUH where this behavior will be changed/can be made [...]

I appreciate the help man, I also just found out that moonraker was having a bug with the file upload that was fixed as of early this morning. I think the main issue I was having was because of that and also me just being very new to this stuff. Thanks for the resources and the reply, and one again, I apologize for creating a thread where it didnt belong.