dw-0/kiauh #202: Permission denied when creating Nginx config for Fluidd

disaster999, 2022-05-30:
### Linux Distribution

Not sure what this is asking, I am installing this on a Orange Pi 3 LTS board running Debian image from Orange Pi's website

### What happened

First time installing klipper on an orangepi, Followed the steps, installed Klipper first, then moonraker, then Fluidd. As it was installed Fluidd, it gets to the point where it is installing Nginx I will get an error saying permission denied

###### Checking for the following dependencies:
● nginx
[✓ OK] Dependencies already met!

###### Creating Nginx configuration for Fluidd ...
/home/orangepi/kiauh/scripts/nginx.sh: line 279: /home/orangepi/kiauh/resources/fluidd: Permission denied

### What did you expect to happen

It should just work?

### How to reproduce

Just following the steps prompted on the screen

th33xitus, 2022-05-30:
Im not quite sure why that permission issue occured.
Though, i pushed a possible fix. I wanted to get my hands on that function anyways, so i just tried a different approach on setting up the config. Please update KIAUH and try again. And please report back if it works now.

disaster999, 2022-05-31:
I tried updating, didnt work. Tried manually installing Fluidd, that was a big fail and kept crashing my board half way. 

I basically formatted everything and start from scratch again. I think I downloaded an outdated Debian image before cuz I downloaded an more updated version for my board, reinstalled everything and went without a hitch. Fluidd works now.

closed 2022-05-31 by disaster999