dw-0/kiauh #203: Error message in "Update Menu"

LifeDay, 2022-05-31:
### Linux Distribution

 Raspbian GNU/Linux 10 (buster) armv7l

### What happened

Update menu loaded much slower than usual and displayed the error message "fatal: Not a valid object name origin/master" just above entry 6 (PrettyGCode)
[image]

### What did you expect to happen

Usually, no error message is displayed

### How to reproduce

1. Launch KIAUH
2. Perform action 2
3. Error message is displayed

th33xitus, 2022-05-31:
Thanks for reporting. Should be solved now.

closed 2022-05-31 by th33xitus

LifeDay, 2022-05-31:
Tried the fix and it looks like it's working now.
Thanks!

commit 8ba46fa4 (2022-05-31): fix(pretty_gcode.sh): use main instead of master (fixes #203)

- pgc uses main instead of master. so checking for origin/master leads to an error