dw-0/kiauh #204: Update All Does not Perform Update

LifeDay, 2022-06-02:
### Linux Distribution

 Raspbian GNU/Linux 10 (buster) armv7l

### What happened

Update menu shows there are updates for Klipper and Moonraker.
Selecting update all does not perform an update but displays the message, "Everything is already up-to-date!"

### What did you expect to happen

I would have expected Klipper and Moonraker to be updated

### How to reproduce

1. Launch Kiauh
2. Select menu **2) [Update]**
3. Menu shows updates for Klipper and Moonraker are available
4. Select option a) [Update all]
5. Message shows "Everything is already up-to-date!"
[image]

### Additional information

The menu is also loading slowly and in a staggered fashion that I don't remember from a week or so ago.

JRHeaton, 2022-06-02:
I'm experiencing the same issue, and also with a multi-instance installation.

th33xitus, 2022-06-03:
i was able to reproduce that issue and im working on a solution. not sure why that broke...

JRHeaton, 2022-06-27:
Still observing this issue in the latest update

commit b07a83c8 (2022-06-30): fix: update all function (#204) (#217)

* fix: update all function (fixes #204)
* fix: replace non-existing `print_unkown_cmd` function
closed 2022-06-30 by th33xitus