dw-0/kiauh #207: Invalid root path (gcodes)

linghaokeji, 2022-06-04:
### Linux Distribution

I tried install klipper+moonraker+fluidd on debian buster and debian bulleyes

### What happened

i installed many times on different systems but it still shows (Invalid root path (gcodes))  

### What did you expect to happen

how to solve that problem

### How to reproduce

install on orangepi's debian system (server version) No error is reported during the process

### Additional information

[image]
[image]

linghaokeji, 2022-06-04:
My English is very poor, if there is anything wrong, please forgive me

th33xitus, 2022-06-04:
You need to make sure that your printer.cfg is valid an Klipper can start without any errors. At that point the error message should disappear.

linghaokeji, 2022-06-04:
ok I will try it later,I will report the progression.thanks

linghaokeji, 2022-06-04:
I tried to connect the mcu board, the g-code upload button is red.it says : root is not available.Please check your logs

th33xitus, 2022-06-04:
Please join the Klipper Discord to get more help: https://discord.klipper3d.org/

This issue is not KIAUH related. And it seems as a general configuration issue.

closed 2022-06-04 by th33xitus