dw-0/kiauh #208: Allow for using scheduled (cron) backups using kiauh scripts

LoadedCamera, 2022-06-06:
### Is your feature request related to a problem? Please describe

Not related to a problem.

### Describe the solution you'd like

I utilize the backup options under the kiauh menu.  This makes a very portable backup that can be pushed to github, or even stored elsewhere.
I am interested in a feature that would let us use the already built-in features of the backup to run via cron.
Having the ability to backup each item we have installed would be very beneficial.
I envision something being put into cron or a personalized script to run manually like this:
./kiauh/scripts/backup.sh klipper
./kiauh/scripts/backup.sh moonraker
./kiauh/scripts/backup.sh moonraker-db
./kiauh/scripts/backup.sh mainsail (or fluidd, based on your installation)
./kiauh/scripts/backup.sh klipperscreen

### Describe alternatives you've considered

I have installed klipper-backup tools and while they work, this is another "tool" to ensure is updated.  Using kiauh, when I update that tool, it would update any backup scripts also.  

### Additional information

Basically, I am looking for an ability to run the included backup.sh for each installed product manually (scripted, etc...)

bitingsock, 2023-07-03:
personally, command line switches would be nice in all the kiauh tools:
`kiauh.sh backup klipper`
`kiauh.sh upgrade klipper`

eloo, 2023-08-24:
hi, i have the same issue/idea right now and came up with the following script so far

this script is placed in my users home directory.. maybe you need to adjust the KIAUH_SRCDIR to your needs

```
#!/bin/bash

### sourcing all additional scripts
#KIAUH_SRCDIR="$(dirname -- "$(readlink -f "${BASH_SOURCE[0]}")")"
KIAUH_SRCDIR="./kiauh"
for script in "${KIAUH_SRCDIR}/scripts/"*.sh; do . "${script}"; done
for script in "${KIAUH_SRCDIR}/scripts/ui/"*.sh; do . "${script}"; done

set_globals

backup_config_dir
backup_moonraker_database
backup_klipper
backup_mainsail
backup_moonraker
backup_klipperscreen
```

but yes maybe such a script could be included in this repository directly

EricZimmerman, 2023-12-17:
consider setting this up, at least for printer configs. you could copy other things to config dir to get them included

https://github.com/EricZimmerman/Voron-Documentation/blob/main/community/howto/EricZimmerman/BackupConfigToGithub.md

closed 2024-03-02 by dw-0