dw-0/kiauh #215: ConditionPathExists in NginX service requires /etc/octopi_streamer/hls

hllhll, 2022-06-25:
### Linux Distribution

Linux octopi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

### What happened

Trying to install Fluidd on a pre-existing OctoPrint installation in an octopi image, Script recommended to remove haproxy and replace it with NginX. I Accepted. Now NginX won't start. Basically killed my existing OctoPrint installation.
```
pi@octopi:~/kiauh $ sudo service nginx status
● nginx.service - NGINX server for serving HLS and JPEG frames
   Loaded: loaded (/lib/systemd/system/nginx.service; enabled; vendor preset: enabled)
   Active: inactive (dead)
Condition: start condition failed at Sat 2022-06-25 20:19:57 BST; 3s ago
           └─ ConditionPathExists=/etc/octopi_streamer/hls was not met
     Docs: man:nginx(8)

Jun 25 20:01:45 octopi systemd[1]: Condition check resulted in NGINX server for serving HLS and JPEG frames being skipped.

```

### What did you expect to happen

Installation would not cause any problems with existing, running, working instance

### How to reproduce

1. Use an octopi image
2. Install Klipper
3. Install (manually) Moonraker
4. Use this script to install Fluidd
5. Try to access any of the interfaces. Moonraker interface works. Both Fluidd and OctoPi are not reachable

hllhll, 2022-06-25:
Fix (for me): Comment out  this next line
```# ConditionPathExists=/etc/octopi_streamer/hls```
Then start NginX 
```sudo service nginx start```

th33xitus, 2022-06-25:
KIAUH doesn't officially support OctoPi images. And i don't plan to support them as they basically come with conflicting software out of the box.

Though, i have never seen this issue and i didn't even know of that NGINX start condition. So in turn i have no idea where to comment out the line you mention.

hllhll, 2022-06-25:
I Wonder myself where this is coming from, Haven't inspected Nginx installation script thoroughly though.
Looks like when NginX is installed/configured something along the pipeline requires a specific "thing" that has to do with octoprint. Maybe this is a NginX that was pre-configured (but not active?) with OctoPi...
Will look this up also in octopi/print

closed 2022-08-15 by th33xitus