dw-0/kiauh #216: octoprint fails to install on ubutnu 20.04 (linux mint)

Nathan22211, 2022-06-25:
### Linux Distribution

Linux mint 20.3, based on ubuntu 20.04

### What happened

upon attempting to install octoprint with the script, it says it's initializing the installation then immediately end the script.ends

### What did you expect to happen

Stuff should show up about the script installing octoprint

### How to reproduce

run the script and attempt to install octoprint. I'm not entirely sure if it's due to my distro or just an issue in your script. So try on ubuntu 20.04 first before trying other distros

Nathan22211, 2022-06-25:
well found out why. You need some checks to tell the user that klipper needs to be installed before installing octoprint

th33xitus, 2022-06-26:
Thanks! I found the issue and pushed a fix for it.

commit 39e22acb (2022-06-26): fix(octoprint.sh): script exiting without error message (fixes #216)

- the script was exiting without notifying the user that klipper has to be installed first
closed 2022-06-26 by th33xitus