dw-0/kiauh #219: autocommit unable to open script

JelleDijkhuizen, 2022-07-05:
### Linux Distribution

MainsailOS

### What happened

I tried to use the autocommit script. When I executed the script via ssh there are no problems but when i use the macro it tells me sh: 0: Can't open /home/pi/backup.sh

### What did you expect to happen

the macro running without a problem

### How to reproduce

create a backup macro with the script in /home/pi/backup.sh

th33xitus, 2022-07-05:
Im unable to give any advice without even knowing the exact content of the macro. Maybe you are experiencing some kind of permission issue here.

th33xitus, 2022-07-05:
Hmmm okay, so its pretty much exactly like written in the wiki article here in this repo.
One thing i notice though... is there really **no** indentation in front of `RUN_SHELL_COMMAND` ?
Afaik indentation is always required after `gcode:`. So each line has to start at least with a single whitespace.
```
[gcode_macro BACKUP_CFG]
gcode:
 RUN_SHELL_COMMAND CMD=backup_cfg
```
Personally i prefer 2-4 whitespaces (usually 4)
```
[gcode_macro BACKUP_CFG]
gcode:
    RUN_SHELL_COMMAND CMD=backup_cfg
```

JelleDijkhuizen, 2022-07-05:
Yeah there are 4 whitespaces i front of the RUN_SHELL

th33xitus, 2022-07-05:
Interesting. I have no other idea right now. Other than maybe trying to move the backup.sh file into your klipper_config folder and try to call that script from there. But i don't really see why it should make a difference. I'll have to try and reproduce that first... i never experienced an issue with that, but then again, i keep that script inside of my klipper_config folder, hence why i came up with that idea right now.

closed 2022-07-18 by th33xitus