dw-0/kiauh #221: When Updating Multi-Instance Moonraker, it fails to start the service

Jericon, 2022-07-11:
### Linux Distribution

Raspbian 11.3

### What happened

Used Kiauh to update Moonraker.  At the start, it stopped the 2 instances that were running.  However at the end it tried to start a single instance, which does not exist.

`###### Restarting Moonraker...
Failed to restart moonraker.service: Unit moonraker.service not found.`

### What did you expect to happen

Start the multiple instances.

### How to reproduce

Update moon raker

th33xitus, 2022-07-11:
Im 95% sure that you saw 2 messages in green stating that the moonraker services were restarted. The one you saw which shows the failing was likely white? Your services were restarted correctly, i am sure.
The message you saw there was from a moonraker-script (external script that KIAUH calls), which happens to report that message when trying to restart moonraker. but it doesn't support restarting of multi instances, hence the error.

I need to get rid of the error messages thrown by the external script i guess

Jericon, 2022-07-11:
Sounds possible. I didn’t check the status of the service after the script ended. I just started it back up. 

If it did show green messages it scrolled past and out of my active window before I saw it.

th33xitus, 2022-07-12:
Okay, i will close this issue in the meantime. If you encounter that issue again, feel free to update this ticket. As of now i can't consider it as a bug.

closed 2022-07-12 by th33xitus