dw-0/kiauh #222: Theme install fails if any folder present in config folder

jatson, 2022-07-13:
### Linux Distribution

Ubuntu 20.04.4 LTS (GNU/Linux 5.4.164-235 armv7l)

### What happened

I have a folder inside my "klipper_config" folder called "archived" to store the old configuration files.
This leads to a failure during automated mainsail theme install: the ".theme" folder will be placed into the subfolder.
Please see the screen attached about the install log:
[image]

### What did you expect to happen

I would expect that the target directory is not any subdirectory inside the "klipper_config" folder but the folder itself

### How to reproduce

- create any sub directory inside the "klipper_config" directory manually. For eg.: "mkdir ~/klipper_config/archived/"
- run the theme installer (Main Menu --> 4) Advanced --> 6) Theme installer)
- install any of the themes (I've tried the "1) [Dracula]" )

th33xitus, 2022-07-20:
Hey, thanks for the report. I just pushed a fix to this issue. Feel free to update KIAUH and confirm if the issue has been resolved. In my tests, the themes now get correctly installed.

closed 2022-07-20 by th33xitus

jatson, 2022-07-22:
Hey, confirmed. Works like charm!
Thanks for the quick fix!

commit ae011963 (2022-07-20): fix: install mainsail theme to correct folder (fixes #222)