dw-0/kiauh #229: Externally installed Klipper detected, then not

bjarchi, 2022-08-15:
### Linux Distribution

Raspbian Linux octopi 5.10.103-v7l+ #1529 SMP Tue Mar 8 12:24:00 GMT 2022 armv7l GNU/Linux

### What happened

Pulled kiauh from git, ran script. Detected previously installed Klipper and Octoprint; octoprint installed via Octopi image and Klipper installed via git clone & klipper docs. 

Tried to use (1) to install Moonraker, but was informed that klipper needed to be installed first. 

Installed moonraker from git following moonraker instructions, but moonraker in kiauh still shows as "incomplete"

Note: both kiaush and Moonraker git repos cloned into ~/code/ instead of directly into ~/

### What did you expect to happen

Klipper properly detected & used for moonraker install/update

Moonraker properly detected when installed externally

### How to reproduce

Install Octopi image; update octoprint & run `sudo apt full-upgrade` 
Install Klipper per instructions on Klipper site

$ mkdir code
$ git clone... kiaush into ~/code/kiaush

Use Kiaush to try to install Moonraker. Fails. Same with upgrade Moonraker (fails trying to run venv moonraker pip)

Install Moonraker per site instructions

Run kiaush; Moonraker still shows as "incomplete"

th33xitus, 2022-08-15:
You likely installed Klipper as an init.d service, which is recognized by KIAUH but not supported anymore. I need to improve the script in that regard. Not in terms of supporting old init.d installations, but to make it more obvious what the real issue is.

Moonraker shows as incomplete because you installed it in a sub-directory. Installations of Klipper or Moonraker in sub-directories of any kind are not supported.

bjarchi, 2022-08-15:
> You likely installed Klipper as an init.d service, which is recognized by KIAUH but not supported anymore. I need to improve the script in that regard. Not in terms of supporting old init.d installations, but to make it more obvious what the real issue is.

That's correct, it's an init.d service configured in `/etc/default/klipper` for the service params. I'm re-learning linux system admin after about a 20yr hiatus, so not up to date on new options. I did however install Klipper from current git master according to current docs, as of a few days ago. Is there a guide for migrating to a more modern service management option without (ideally) reinstalling klipper? I suppose I could reinstall through `kiauh` since all the meat is in my configs, assuming raspbian is supported. 

> Moonraker shows as incomplete because you installed it in a sub-directory. Installations of Klipper or Moonraker in sub-directories of any kind are not supported.

Is this something you'd consider as a feature request, maybe an optional `prefix` for externally installed moonraker and etc.? I find it keeps things clean to keep repo checkouts in a subdir off `$HOME`. 

If not I guess options are 

1. Move moonraker checkout and reinstall
2. symlink moonraker checkout directory into `~/` ... that should make KIAUH see it without needing to reinstall? 

Thanks!

th33xitus, 2022-08-15:
> I did however install Klipper from current git master according to current docs, as of a few days ago.

The official Klipper docs are, especially in terms of installing Klipper, kind of outdated.

> Is there a guide for migrating to a more modern service management option without (ideally) reinstalling klipper?

No other guide other than removing your currently installed Klipper service and re-install with KIAUH. KIAUH can uninstall your currently installed SysVinit installation.

> Is this something you'd consider as a feature request, maybe an optional prefix for externally installed moonraker and etc.?

Currently not, no. Although i see the point in keep things clean, that is currently something im not really interested in implementing. Maybe in the future...

bjarchi, 2022-08-15:
> The official Klipper docs are, especially in terms of installing Klipper, kind of outdated.

Too bad. I think I noticed some other things that need updating... maybe I'll look into updating some of that. IIRC it's largely generated from files in the repo

> No other guide other than removing your currently installed Klipper service and re-install with KIAUH. KIAUH can uninstall your currently installed SysVinit installation.

Fair enough. Given that most of the setup is in the configs, this actually seems like a pretty painless process... Although I may look for a backup SD card before I do so. 

> Currently not, no. Although i see the point in keep things clean, that is currently something im not really interested in implementing. Maybe in the future...

OK. Maybe I'll look into supporting a PR when free time allows. In the meantime I'll probably start fresh, installing from KIAUH. Thanks for the replies.

closed 2022-08-15 by bjarchi