dw-0/kiauh #230: Install moonraker fails: Klipper not installed. Install klipper fails: service already installed.

aeiounix, 2022-08-17:
### Linux Distribution

Octopi 

### What happened

http://paste.c-net.org/r82nycxbpxvh

Moonraker installation fails with error about klipper being absent. 
Klipper installation fails with message about klipper being present already. 

### What did you expect to happen

Expected moonraker installation.

### How to reproduce

Run the script. 
Select option 1. 
Select option 2. 
Select option 1. 

### Additional information

This is on an install where octoprint has been in use for some time. Recently converted to klipper. 
First time trying to install mainsail or fluidd.

th33xitus, 2022-08-17:
You have a rather old Klipper installation on that OctoPi system.
```
[2022-08-17 16:14:36] [ERR]: Moonraker setup started without Klipper being installed. Aborting setup.
[2022-08-17 16:16:04] [INFO]: Found Klipper service: /etc/init.d/klipper
```
I need to improve the feedback the user is given while running the script. Currently it seems to be confusing, i agree.

Your solution would be, to run the Klipper uninstaller with KIAUH (KIAUH can remove those old service installations) and then re-install Klipper with KIAUH. Then everything should work.

simonachmueller, 2022-09-07:
I recently installed klipper as in documentation https://www.klipper3d.org/Installation.html, still got same error. Reinstalled using script and it seems to work.

th33xitus, 2022-09-08:
That is correct and intended. You will always get the error if Klipper is installed as SysVinit service. It is not supported by KIAUH.
But you should have been shown a new, and better error message in KIAUH saying something like:
`Unsupported Klipper SysVinit service detected ... Please re-install Klipper with KIAUH!`

I will further improve the error feedback and try to display it in the moonraker setup as well, so users don't have to execute the Klipper installer first to see that error.

simonachmueller, 2022-09-10:
Btw maybe you can collaborate with klipper community and update klipper installation [documentation](https://www.klipper3d.org/Installation.html) to make first-comers aware that your script is actually more straightforward way to get it working rather then using octopi image and struggle installing Mainsail etc on it.

closed 2022-10-08 by aeiounix