dw-0/kiauh #234: [FR] Add CanBoot to install menu

Markus-1984, 2022-08-20:
### Is your feature request related to a problem? Please describe

Please add CanBoot install option.

### Describe the solution you'd like

Add CanBoot install menu point

th33xitus, 2022-08-24:
Will be hard for me to test since i don't own any CanBoot capable hardware. I usually dislike to implement stuff i can't properly test myself as im simply not able to support anything at all if it comes to any issues with the script.

tome9111991, 2022-10-03:
maybe only a option to clone canboot to home folder?

EricZimmerman, 2023-12-17:
this is essentially git cloning Katapault (the new canboot name)