dw-0/kiauh #236: WSL support

dmason1992, 2022-08-23:
### Is your feature request related to a problem? Please describe

I am trying to install everything using kiauh within a WSL installation.

I've managed to expose the USB port.

Kiauh install scripts seem to work (although possible I missed errors that don't prevent script finishing)

Systemd isn't part of the Linux distributions for WSL and this seems to be the route of the problem. I managed to launch ngnix which allowed mainsail to run.  Then I launched moonraker manually and it showed as communicating on the mainsail UI.

This was was far as I've got so far. But I guess my question is - is there a known alternative we can use instead of Systemd for WSL? And how hard would it be to detect you are installing on WSL and use that method instead?

Cheers

### Describe the solution you'd like

Kiauh to detect it is installing in a WSL environment. Use some other method other than Systemd to launch the 3 components.

Use case for me as a reviewer is I can do lots of setup and testing of control boards from my pc without setting up an SBC etc. / gives people an option to run klipper from Windows pcs. 

### Additional information

Installing on Ubuntu due to guides existing for exposing usb ports. Same issue with Systemd happens on debian as well though as I checked.

th33xitus, 2022-08-24:
Honestly im not interested in supporting WSL, i rather expect WSL to support systemd instead the other way around. If that's not going to happen, no official KIAUH support for that. And even then, im not sure if WSL should be used for controlling a 3D printer at all. 

You can hack systemd into WSL though. I did that myself in the past but the guide changed a bit since: https://github.com/shayne/wsl2-hacks

Another way could be, to set up a VM in VMware or similar and just pass through the controller boards via USB. That should work, as this is something i already did.

dmason1992, 2022-08-24:
Well thank you for pointing me in the right direction! I got there in the end using the github repo you linked! Kiauh installation works using that method (although i need to start from the beginning again and do the hack first before the install as currently i need to manually launch the 3 services still on startup)

th33xitus, 2022-08-25:
You could try and run `sudo systemctl enable klipper.service` to enable the automatic start on boot for that service. Use the same command for the other two services. Otherwise, just reinstalling also takes only about a minute.

dmason1992, 2022-08-25:
Well I'll try those commands first :) thank you again!

closed 2022-08-25 by th33xitus