dw-0/kiauh #238: Command python with Asterisk * in the path

Travis90x, 2022-09-02:
### Linux Distribution

Debian 11 (bullseye)
Kernel: Linux 5.15.27-station
Architecture: arm64
Klipper v0.10.0-557
Mainsail v2.2.1 

### What happened

```
Traceback (most recent call last):
File "/home/pi/klipper/scripts/calibrate_shaper.py", line 174, in 

main()
File "/home/pi/klipper/scripts/calibrate_shaper.py", line 153, in main
datas = [parse_log(fn) for fn in args]
File "/home/pi/klipper/scripts/calibrate_shaper.py", line 153, in

datas = [parse_log(fn) for fn in args]
File "/home/pi/klipper/scripts/calibrate_shaper.py", line 19, in parse_log
with open(logname) as f:
FileNotFoundError: [Errno 2] No such file or directory: '/tmp/resonances_x_*.csv'
```

### What did you expect to happen

`/tmp/resonances_x_*.csv`
the asterisk ->`*` have to let to choose all the file like
resonances_x_2022_09_02.csv
resonances_x_2022_09_01.csv

### How to reproduce

[gcode_macro calibrate_x]
gcode:
    RUN_SHELL_COMMAND CMD=shaper_calibrate_x

[gcode_shell_command shaper_calibrate_x]
    command: ~/klipper/scripts/calibrate_shaper.py /tmp/resonances_x_*.csv -o /tmp/shaper_calibrate_x.png
    timeout: 30.
    verbose: True

### Additional information

The command works fine in terminal
~/klipper/scripts/calibrate_shaper.py /tmp/resonances_x_*.csv -o /tmp/shaper_calibrate_x.png

th33xitus, 2022-09-02:
It's likely that the command in the shell command is not getting expanded. meaning wildcard characters have no effect. just execute a shell script and not directly the command.

closed 2022-09-02 by th33xitus

Travis90x, 2022-09-12:
I can execute a bash with that command correctly with
 command: sh /home/pi/klipper/scripts/generate-shaper-graph-x.sh
 scripts here
 https://github.com/Travis90x/TwoTrees-Sapphire-Plus-SP5/tree/default/Klipper-Firmware/scripts

but in the klipper macro I can't use:

[gcode_shell_command shaper_calibrate_x]
 command: sh ~/klipper/scripts/generate-shaper-graph-x.sh
or 
 command: sh /home/$USER/klipper/scripts/generate-shaper-graph-x.sh

th33xitus, 2022-09-12:
likely because you **again** use a part in the command which need to be expanded.
`~` and `$USER` in that case.