dw-0/kiauh #243: Crowsnest Install

mlee12382, 2022-10-09:
### Is your feature request related to a problem? Please describe

Had to install crowsnest separately for multi-cam

### Describe the solution you'd like

Include Crowsnest install either as a separate option or rolled into one of the other installs.

th33xitus, 2022-10-10:
Yes it is planned, i just haven't found time for implementing it. The plan is to replace mjpg-streamer with crowsnest and just keep the mjpg-streamer installer as an alternative.

mlee12382, 2022-10-11:
Another thing that would be useful is the installation of the raspi mcu files to control the gpio with klipper.

th33xitus, 2022-10-11:
That's also on my todo :D

Enderlesss, 2022-12-07:
Hello th33xitus, I was hoping you could help me with an issue, I installed crowsnest after installing tailscale and 2 instances of klipper.

I keep getting make: *** [Makefile:37: install] Error 1

i have tried unistalling with the same error, i am clueless now. Can you point me in the right direction 

thanks,

th33xitus, 2022-12-07:
@Enderlesss It's probably the best, if you join the Mainsail Discord and open a new thread in the support forum. In there, your issue can be reviewed and in case there is a bug with Crowsnest, you'll likely get new instructions on how to handle that (maybe creating an issue in the Crowsnest Issue tracker or maybe the Crowsnest-Developer will write it on his Todo-List).

closed 2023-02-03 by th33xitus