dw-0/kiauh #244: new moonraker update has different needs of configurations

ramonskie, 2022-10-16:
### Linux Distribution

armbian

### What happened

when installing
- klipper
- moonraker
- fluidd

i do not see any config files. as location is now in `./printer_data/config/`
also some config options are now deprecated.

### What did you expect to happen

a default configuration set for moonraker and printer macros etc

### How to reproduce

fresh install

d-nx, 2022-10-16:
moonraker.conf also needs to be updated. The current kiauh moonraker install creates a moonraker.conf file with depreciated config sections that prompt errors.

th33xitus, 2022-10-16:
yes i am aware and i am already working on the changes, im just not so fast 😄 and i want to make sure i don't push garbage code to the repo

ramonskie, 2022-10-17:
yeah i just happened to run it to it when i did a complete reinstall..
so just wanted to give a headsup.. no rush here as i got it working manually..

romanlex, 2022-10-17:
@ramonskie can you prodive how did it manually?

ramonskie, 2022-10-17:
i ran `./kiauh.sh` choose the settings option
change the folder from `/home/username/klipper_config`
to `/home/username/printer_data/config`
copy klipper_config to the printer_data/config
also change the the path to `gcode_files` to `printer_data/gcodes`

i removed the deprecated settings from moonraker.conf.but for some reason it still shows up as a warning.
but the printing works fine from this point on.

cristiantrif, 2022-10-17:
maybe this change from klipperWrt repo will help https://github.com/ihrapsa/KlipperWrt/commit/31e837c0f748d06f9f27242db81d9396d180053b

cristiantrif, 2022-10-17:
@th33xitus hope it helps

th33xitus, 2022-10-17:
@cristiantrif thanks for the link. i already made those changes to kiauh locally. the "problem" is, that kiauh has become much more complex over the time, so i have several other spots to change if i don't want to partially disable functions for the meantime (which i actually don't intend to but maybe its necessary).

cristiantrif, 2022-10-17:
@th33xitus maybe is easier to use an older version of moonraker for now

ramonskie, 2022-10-17:
> @th33xitus maybe is easier to use an older version of moonraker for now

the only problem with this is all the auto update scripts for this.+
the moonraker update should just be considered a breaking change from this point

th33xitus, 2022-10-17:
Personally, i consider it a breaking change, yes. I will also increase KIAUHs major version to version 5 once im ready to push the changes. I won't implement functions to convert old/legacy setups to the new file structure introduced by Moonraker. It's just way too much work and re-installing all components will be much faster in the end.

cristiantrif, 2022-10-17:
@th33xitus sounds good for me

romanlex, 2022-10-17:
@ramonskie thx for the instruction
but for me I'm still getting the error "Moonraker requires sudo permission to update the system service. Please check your notifications for further intructions." ((( 
I can't edit any files with fluidd

Travis90x, 2022-10-17:
> @ramonskie thx for the instruction but for me I'm still getting the error "Moonraker requires sudo permission to update the system service. Please check your notifications for further intructions." ((( I can't edit any files with fluidd

try with this port
http://192.168.0.XXX:7125/
give password to moonraker
if your user is named "pi"
sudo chown -R pi /home/pi/printer_data

tficken, 2022-10-18:
> i ran `./kiauh.sh` choose the settings option change the folder from `/home/username/klipper_config` to `/home/username/printer_data/config` copy klipper_config to the printer_data/config also [...]

+1 on this working

eV1lMoF0, 2022-10-18:
I got it running without error and without using a password, have not printed anything yet but it reads the printer and sends commands to the printer like home and move x, y, z

what I did was take the printer.cfg that is in the ~/klipper_config/ directory and modified it like this

1. [include mainsail.cfg]
2. [include /home/USERNAME/printer_data/config/printer.cfg]

then I put all my printers config in the /home/USERNAME/printer_data/config/printer.cfg file

Dont know if this will help anybody, just a quick and dirty way I got it working till they get things worked out

ramonskie, 2022-10-18:
> > try with this port http://192.168.0.XXX:7125/ give password if moonraker request it if your user is named "pi" sudo chown -R pi /home/pi/printer_data
> 
> I'm trying this several times any passwords are not accepted(from user, form root
so the user you used to run/install kiauh should have sudo rights
and use that users password in the moonraker_url:7125

Travis90x, 2022-10-18:
I started with a clean ARMBIAN system (debian ARM on a Android TV Box, it's similar to Raspberry OS)
installed Kiauh and than I proceded in this way
[image]

romanlex, 2022-10-18:
maybe who known how set sudo password for moonraker in ubuntu?
[image]
[image]
[image]

romanlex, 2022-10-18:
omg 
in my system doesn't exist lost+found directory
[image]
`sudo mkdir /lost+found` resolve issue
[image]

ramonskie, 2022-10-19:
nice find.
but why would the code depend on `/lost+found` @th33xitus

th33xitus, 2022-10-19:
@ramonskie you're asking the wrong person. that is moonraker source code. i can't answer that question.

XoBoR, 2022-10-19:
Is there an ETA for the new KIAUH version?

marbocub, 2022-10-19:
I’ｍ very interested in this topic for multiple intances.
How do you plan filenames and directory structures?

for example:
~/printer_data/config/printer_1.cfg
or
~/printer_1_data/config/printer.cfg

th33xitus, 2022-10-19:
> How do you plan filenames and directory structures?

It will definitely be in the form of `~/printer_1_data/config/printer.cfg`. This file/folder-structure will be easier to handle in the long-term.

No ETA so far. I hope i can get something published this week, maybe a temporarily slimmed down version of KIAUH at least where the most important components like Klipper, Moonraker, Mainsail and Fluidd completely work and the other ones are temporarily disabled until i caught up and applied the changes to them as well. It will probably also include some talking to other devs, especially for Obico and the Telegram Bot.

aboktor, 2022-10-20:
What is the possibility of pinning the version of moonraker to the last known good version until a fix is worked on? May be something to checkout a specific version of moonraker.

th33xitus, 2022-10-20:
I just merged the following PR: https://github.com/th33xitus/kiauh/pull/245

Please ALL read the changelog carefully: https://github.com/th33xitus/kiauh/blob/v5.0.0/docs/changelog.md#2022-10-20

Travis90x, 2022-10-21:
For now Kiauh v5 doesn't support everything, so with a clean OS

```
cd ~
rm -rf ./kiauh
git clone https://github.com/th33xitus/kiauh.git
./kiauh/kiauh.sh
```

install Klipper with python 3, 1 instance, moonraker, mainsail, fluidd port 81.
backup kiauh 5
`mv ./kiauh ./kiauh5`

install Kiauh 4
```
git clone https://github.com/th33xitus/kiauh.git -b v4
./kiauh/kiauh.sh
```
install Klipperscreen, PrettyGCode, MJPEG-Streamer, Gshell command

return to kiauh 5
```
mv ./kiauh ./kiauh4
mv ./kiauh5 ./kiauh
```

If Klipperscreen or others recreate old directories (ex. "klipper_config")
```
mv -v ~/klipper_config/* ~/printer_data/config/
mv -v ~/klipper_logs/* ~/printer_data/logs/
rm -rf ~/klipper_config
rm -rf ~/klipper_logs
sudo ln -s /home/pi/printer_data/config /home/pi/klipper_config
sudo ln -s /home/pi/printer_data/logs /home/pi/klipper_logs
```

If there where Webcamd issues:
with Kiauh4 
```
mv ./kiauh ./kiauh5
mv ./kiauh4 ./kiauh
./kiauh/kiauh.sh
```
remove MJPG-Streamer
reinstall MJPG-Streamer.
return to kiauh 5 (see above)

All tested with Armbian on TV Box Allwinner H6 (T95 max) and webcam Microdia N5 (Sonix_Technology).

mirokymac, 2022-10-21:
> I just merged the following PR: #245
> 
> Please ALL read the changelog carefully: https://github.com/th33xitus/kiauh/blob/v5.0.0/docs/changelog.md#2022-10-20

Just add a path patch script for V5.0.0.

For machines that installed V4.0.0, just pull update with `kiauh.sh`, then run `path_patch_4to5.sh` will move files accroding `klipper service instance names` to new file structure.

Before patch, `.moonraker_database` and `klipper_config` folders are backup to `~/kiauhV4_config_backup`.

Tested on my 4 machines.

th33xitus, 2022-10-21:
Sorry, as i already said in an earlier post in this issue, im not really interested in implementing functions for patching/migrating old installs. I seem to see a few flaws in your script also (nothing serious though), so feel free to use them if they work for you. Im likely not going to merge that. Sorry.

lokertr, 2022-10-22:
Has anyone had any luck with Xubuntu? My install got fouled up from the update so I wiped the system to start fresh. Even doing that though Moonraker complains that the directory Klipper is requesting gcodes from does not match what it expects.   It demands I update it to /home/travis/printer_data/gcodes...tried updating Virtual SDCard in mainsail.cfg.... tried adding it to printer.cfg but no change. I am just currently unable to open any gcode files.

larie438, 2022-10-29:
Do you have a rough timeline of when KlipperScreen will be available again?

th33xitus, 2022-10-29:
The installer is available again. Im not 100% sure if everything works out of the box though. KIAUH only executes KlipperScreens own install script, so actually disabling it wasn't really necessary in the first place, i just wanted to have a look at it first when i did that. But again, not 100% sure if maybe some manual intervention is required now. It might be.

closed 2023-01-19 by th33xitus