dw-0/kiauh #256: Wrong ordering when installing more than 10 instances

XoBoR, 2022-11-01:
### Linux Distribution

not relevant

### What happened

When you install more than 10 instances, due to alphanumeric sorting printer 10 becomes the first instance on port 7125, printer 11 is on 7126 and so on.

### What did you expect to happen

ordering by instance number instead of the name of the printer

### How to reproduce

install more than 10 instances

### Additional information

Not a big issue but it'd still be nicer if this was fixed.

th33xitus, 2022-11-01:
Okay that would indeed make sense in case of "printer_<num>" as instance names. Im just thinking about how to deal with it when there is no index in the name itself, so in case of custom names. I will think about it, but your proposal actually makes sense in the mentioned scenario, yes.

dw-0, 2024-08-31:
This should be fixed with KIAUH v6 which's alpha got released just a minute ago.

closed 2024-08-31 by dw-0