dw-0/kiauh #258: Install Mainsail or Fluidd without Moonraker

marbocub, 2022-11-04:
### Is your feature request related to a problem? Please describe

I am constructing a multi host Klipper environment by a backend host for multi instance Klippy and Moonraker, and a web only host for Mainsail and Fluidd.

I want to use KIAUH to install Mainsail or Fluidd, however, KIAUH need installed Moonraker before installing Mainsail or Fluidd. The Moonraker is unnessesary to install for the web only host.

### Describe the solution you'd like

Do you have a plan to enable installing Mainsail or Fluidd without Moonraker?

### Describe alternatives you've considered

May I send PR #247 or a similar improved version?

th33xitus, 2022-11-04:
Hi!
Sorry for the late reply, i actually planned to give you feedback in your PR. I just haven't had much time lately due to real life. Now i see that you already closed your PR, should i re-open it? I think the idea is good. Actually a thing a haven't considered yet that people want to do that. But it seems possible.

th33xitus, 2022-11-04:
I briefly looked at the PR. I don't think it's required to change anything in regards to NGINX. Mainsail and (afaik) Fluidd both allow to connect to external Moonraker instances. So i think the only necessary step is to remove the restriction in KIAUH and in case of Mainsail, for instance storage the config.json has to be modified to use "browser".

th33xitus, 2023-06-06:
Resolved by https://github.com/th33xitus/kiauh/pull/347