dw-0/kiauh #260: Installation instructions missing vital info

apphead, 2022-11-05:
### Is your feature request related to a problem? Please describe

I tried /home/pi/klipper and it just would not connect. Since there is zero feedback in the error message (mcu not found) I wasted 1 hour barking up the wrong tree, suspecting my printer.cfg to be faulty. Glad it was mentioned _to move the files to their new location_ in the changelog somewhere....

### Describe the solution you'd like

Please add path to config (usually /home/pi/printer_data/config ) on main github page at the end of section _Instructions_. I hope / bet I'm not the only one having had this issue (before).

th33xitus, 2022-11-05:
Those installation instructions you request have nothing to do with KIAUH. I don't see a point in adding such instructions to the main github page as it has nothing to do with KIAUH itself. Thats not the place for other installation instructions. Also, those locations are usually displayed upon creation in the terminal. Maybe i can make it more obvious inside of KIAUH. But usually people just overlook any info message, no matter how strikingly obvious i try to make them.

closed 2022-11-05 by th33xitus

apphead, 2022-11-06:
Must have overlooked the instructions during creation. My bad. But it's far from obvious to use printer_data folder for the config files (and specific to KIAUH setup too). Can't hurt to include that info to the main page IMHO.