dw-0/kiauh #265: Internal Error when trying to update Klipper

menthol1979, 2022-11-14:
### Linux Distribution

Operating System: Raspbian GNU/Linux 10 (buster)
Kernel: Linux 5.10.103-v7l+
Architecture: arm

### What happened

When I attempted to update Klipper, solo or along with other updates, I get this message: 

/=======================================================\
|     ~~~~~~~~~~~~~~~~~ [ KIAUH ] ~~~~~~~~~~~~~~~~~     |
|        Klipper Installation And Update Helper         |
|     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~     |
\=======================================================/

###### Stop klipper.service ...
[âœ“ OK] Stop klipper.service successfull!

###### Updating Klipper ...
Already up to date.

###### Reading dependencies...
>>>>>> Internal Error: missing parameter 'python_version' during function call of install_klipper_packages

### What did you expect to happen

Klipper is reported as updated after this, but I doubt since the update process was never performed.

### How to reproduce

Update all via KIAUH

th33xitus, 2022-11-14:
I was able to confirm that issue. The git pull was done, so your Klipper is now actually up2date.
It's failing after the git pull, when it tries to install possible new dependencies for Klipper.

menthol1979, 2022-11-14:
Oh then that's fine, Klipper side. I'll halt any further updates until it is resolved, in case I fall into a nasty dependency. 
Thanks for your time and efforts! If you need any additional info, like logs or whatever, let me know.
Keep it up, we love it! 
-nick.

th33xitus, 2022-11-14:
I just pushed a fix for that issue. Update KIAUH and try updating again. KIAUH will report an up2date Klipper still, but you can run the update anyways. You should not get any error anymore and see the script actually checking dependencies and all that stuff that should normally happen.

menthol1979, 2022-11-14:
Just updated. Curiously, it didn't report an update to Klipper, but the manual update seemed to work fine (it did not show the red message and checked dependencies normally). Plus, Klipper was restarted fine (previously it remained down after the red message). So, I guess, case closed! Thanks for all your great work!