dw-0/kiauh #267: fluidd zip file location has moved

DrieStone, 2022-11-23:
### Linux Distribution

Raspberry Pi

### What happened

Running the Fluidd installer I get a 404 error.
[image]

### What did you expect to happen

I would expect it could download the package and install.

### How to reproduce

Run the installer and choose Fluidd

DrieStone, 2022-11-23:
It's not the location. It looks like the version number isn't being pulled down at all. I manually put the version number into the stable release string and it worked.

th33xitus, 2022-12-07:
Im not able to reproduce the issue. I just tried it myself to verify and installing Fluidd v1.22.0 worked without any issues.
[image]